YosysHQ/yosys #50: BRAM initialization

ferrandi, 2015-02-10:
I'm trying to synthesize with yosys Verilog code instantiating a standard BRAM template.
Yosys works fine until I've added the BRAM initialization.
The snipped of code creating problems is the following:
  initial
  begin
    $readmemb(INIT_file, memory, 0, n_elements-1);
  end

Is there any plan for the support of BRAM initialization?

Cheers,

Fabrizio

cliffordwolf, 2015-02-10:
Yes. Imo the lack of support for BRAM initialization is one of the biggest showstoppers atm for FPGA flows. I am aiming to have BRAM initialization support in Yosys 0.6, i.e. some time within the next few months, probably sooner if you follow the git head.

cliffordwolf, 2015-02-10:
Just for clarification: The $readmemb is supported in Yosys. It will just prohibit bram instantiation and thus the memory will be implemented with logic gates and flip-flops.

Do you have a problem in read_verilog with $readmemb (syntax error, etc)?

ferrandi, 2015-02-11:
No, I do not have problems with $readmemb. Anyway, I've got the impression that it is better to pass -nomem2reg to read_verilog. This seems to avoid some problems with multi-port inference. Is my feeling supported by some facts?

For the sake of clarity the true script I've used is a tcl script passed to vivado 2014.4 where yosys has been used as RTL synthesizer. Here it is the script:
```tcl
set outputDir HLS_output//Synthesis/vivado_flow
file mkdir $outputDir
create_project main_minimal_interface -force
chan configure stdout -buffering none
exec >&@stdout yosys -p "read_verilog -nomem2reg -defer top.v" -p "synth_xilinx -edif HLS_output//Synthesis/vivado_flow/main_minimal_interface.edif -top main_minimal_interface"
read_xdc HLS_output//Synthesis/vivado_flow/main_minimal_interface.sdc
read_edif $outputDir/main_minimal_interface.edif
link_design -top main_minimal_interface -part xc7z020clg484-1
write_checkpoint -force $outputDir/post_synth.dcp
report_timing_summary -file $outputDir/post_synth_timing_summary.rpt
report_utilization -file $outputDir/post_synth_util.rpt
opt_design
report_utilization -file $outputDir/post_opt_design_util.rpt
place_design -directive Explore
report_clock_utilization -file $outputDir/clock_util.rpt
# Optionally run optimization if there are timing violations after placement
if {[get_property SLACK [get_timing_paths -max_paths 1 -nworst 1 -setup]] < 0} {
  puts "Found setup timing violations => running physical optimization"
  phys_opt_design
}
write_checkpoint -force $outputDir/post_place.dcp
report_utilization -file $outputDir/post_place_util.rpt
report_timing_summary -file $outputDir/post_place_timing_summary.rpt
route_design -directive Explore
write_checkpoint -force $outputDir/post_route.dcp
report_route_status -file $outputDir/post_route_status.rpt
report_timing_summary -file $outputDir/post_route_timing_summary.rpt
report_power -file $outputDir/post_route_power.rpt
report_drc -file $outputDir/post_imp_drc.rpt
report_utilization -file $outputDir/post_route_util.rpt
close_design
close_project
```

cliffordwolf, 2015-02-11:
> Anyway, I've got the impression that it is better to pass -nomem2reg to read_verilog. This seems to avoid some problems with multi-port inference. Is my feeling supported by some facts?

The mem2reg mechanism detects cases that cannot be implemented using the current yosys memory model and transforms those memories to registers directly in the verilog front-end without going through the memory representation. If you call the front-end with -nomem2reg (or set the nomem2reg attribute on a module or array), then you disable this mechanism and force usage of the memory representation. This can (and usually will) result in simulation synthesis mismatches.

From the README:

    - The "nomem2reg" attribute on modules or arrays prohibits the
      automatic early conversion of arrays to separate registers. This
      is potentially dangerous. Usually the front-end has good reasons
      for converting an array to a list of registers. Prohibiting this
      step will likely result in incorrect synthesis results.

For example:

    module test (input clk, a, output reg b);
      reg mem [3:0];
      always @(posedge clk) begin
      	mem[0] = a;
        b = mem[0];
      end
    endmodule
 
This module will introduce one clock cycle delay from `a` to `b` when synthesized without -nomem2reg (which is correct) and two clock cycles delay when synthesized with -nomem2reg (which is incorrect). You should only use -nomem2reg when you are sure that the mem2reg mechanism did incorrectly determine that the conversion to registers was necessary.

If you have a case where mem2reg triggers incorrectly (i.e. -nomem2reg produces correct (and hopefully also better) results because the memory can in fact be described with the current yosys memory model), then I'd be interested in a test case.

ferrandi, 2015-02-13:
I've a testcase where a module inferring a BRAM with initialization is not properly synthesized in flip-flops and gates.
Vivado after reading the netlist during the opt_design command (e.g., Phase 3 Sweep) stop the synthesis since it founds Multi-driven nets. The same verilog is correctly synthesized by vivado with synth_design command

Here in the following the verilog code:
```verilog
// File automatically generated by: PandA framework version=0.9.3-dev
// Send any bug to: panda-info@elet.polimi.it
// ************************************************************************
// The following text holds for all the components tagged with PANDA_GPLv3.
// 
// This hardware description is free; you can redistribute it and/or modify
// it under the terms of the GNU General Public License as published by
// the Free Software Foundation; either version 3, or (at your option)
// any later version.
// 
// This hardware description is distributed in the hope that it will be useful,
// but WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY
// or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License
// for more details.
// 
// You should have received a copy of the GNU General Public License
// along with the PandA framework; see the files COPYING
// If not, see <http://www.gnu.org/licenses/>.
// ************************************************************************

`ifdef __ICARUS__
  `define _SIM_HAVE_CLOG2
`endif
`ifdef VERILATOR
  `define _SIM_HAVE_CLOG2
`endif
`ifdef MODEL_TECH
  `define _SIM_HAVE_CLOG2
`endif
`ifdef VCS
  `define _SIM_HAVE_CLOG2
`endif
`ifdef NCVERILOG
  `define _SIM_HAVE_CLOG2
`endif
`ifdef XILINX_SIMULATOR
  `define _SIM_HAVE_CLOG2
`endif
`ifdef XILINX_ISIM
  `define _SIM_HAVE_CLOG2
`endif

// This component is part of the BAMBU/PANDA IP LIBRARY
// Copyright (C) 2004-2015 Politecnico di Milano
// Author(s): Fabrizio Ferrandi <fabrizio.ferrandi@polimi.it>
// License: PANDA_GPLv3
`timescale 1ns / 1ps
module ADDRESS_DECODING_LOGIC(clock, reset, in1, in2, in3, sel_LOAD, sel_STORE, S_oe_ram, S_we_ram, S_addr_ram, S_Wdata_ram, Sin_Rdata_ram, S_data_ram_size, Sin_DataRdy, out1, Sout_Rdata_ram, Sout_DataRdy, proxy_in1, proxy_in2, proxy_in3, proxy_sel_LOAD, proxy_sel_STORE, proxy_out1, dout_a, dout_b, memory_addr_a, memory_addr_b, din_value_aggregated, be, bram_write);
  parameter BITSIZE_in1=1, BITSIZE_in2=1, BITSIZE_in3=1, BITSIZE_out1=1, BITSIZE_S_addr_ram=1, BITSIZE_S_Wdata_ram=8, BITSIZE_Sin_Rdata_ram=8, BITSIZE_Sout_Rdata_ram=8, BITSIZE_S_data_ram_size=1, address_space_begin=0, address_space_rangesize=4, BUS_PIPELINED=1, BRAM_BITSIZE=32, PRIVATE_MEMORY=0, USE_SPARSE_MEMORY=1, HIGH_LATENCY=0, BITSIZE_proxy_in1=1, BITSIZE_proxy_in2=1, BITSIZE_proxy_in3=1, BITSIZE_proxy_out1=1, BITSIZE_dout_a=1, BITSIZE_dout_b=1, BITSIZE_memory_addr_a=1, BITSIZE_memory_addr_b=1, BITSIZE_din_value_aggregated=1, BITSIZE_be=1, nbit_read_addr=32, n_byte_on_databus=4, n_mem_elements=4, n_bytes=4;
  // IN
  input clock;
  input reset;
  input [BITSIZE_in1-1:0] in1;
  input [BITSIZE_in2-1:0] in2;
  input [BITSIZE_in3-1:0] in3;
  input sel_LOAD;
  input sel_STORE;
  input S_oe_ram;
  input S_we_ram;
  input [BITSIZE_S_addr_ram-1:0] S_addr_ram;
  input [BITSIZE_S_Wdata_ram-1:0] S_Wdata_ram;
  input [BITSIZE_Sin_Rdata_ram-1:0] Sin_Rdata_ram;
  input [BITSIZE_S_data_ram_size-1:0] S_data_ram_size;
  input Sin_DataRdy;
  input [BITSIZE_proxy_in1-1:0] proxy_in1;
[... 1137 lines cut ...]
  puts $ofile_report "      <item stringID=\"XILINX_DESIGN_DELAY\" value=\"$design_delay\"/>"
  puts $ofile_report "    </section>"
  puts $ofile_report "  </application>"
  puts $ofile_report "</document>"
  close $ofile_report
}; #END PROC
set outputDir vivado_flow
file mkdir $outputDir
create_project STD_BRAM_wrapper -force
chan configure stdout -buffering none
exec >&@stdout yosys -p "read_verilog -defer STD_BRAM.v" -p "synth_xilinx -edif vivado_flow/STD_BRAM_wrapper.edif -top STD_BRAM_wrapper"
read_xdc STD_BRAM_wrapper.sdc
read_edif $outputDir/STD_BRAM_wrapper.edif
link_design -mode out_of_context -top STD_BRAM_wrapper -part xc7z020clg484-1 
set_property HD.CLK_SRC BUFGCTRL_X0Y16 [get_ports clock]
dump_statistics
opt_design
dump_statistics
place_design
dump_statistics
route_design
dump_statistics
close_design
close_project
```

cliffordwolf, 2015-02-13:
The following (much smaller ;) testcase shows what is going on here:

	module mem2reg_with_two_always_blocks(
		input clk,
		input [1:0] a_addr, a_din, b_addr, b_din,
		input a_wen, b_wen,
		output reg [1:0] a_dout, b_dout
	);
		reg [1:0] memory [0:3];

		initial begin
			memory[0] = 0;
			memory[1] = 1;
			memory[2] = 2;
			memory[3] = 3;
		end

		always @(posedge clk) begin
			if (a_wen)
				memory[a_addr] <= a_din;
			a_dout <= memory[a_addr];
		end

		always @(posedge clk) begin
			if (b_wen)
				memory[b_addr] <= b_din;
			b_dout <= memory[b_addr];
		end
	endmodule

After mem2reg this are two always blocks each inferring a flop flop to drive each element of memory[].

Interestingly if I synthesize above example with vivado, it will produce a RAMB18E1 cell. (For a total of 8 bits of memory!!!) So even vivado has its problems fully understanding what is going when you write to memory from two always blocks. The following implementation using only one always block will produce the expected result (with both vivado and yosys):

	module mem2reg_with_two_always_blocks(
		input clk,
		input [1:0] a_addr, a_din, b_addr, b_din,
		input a_wen, b_wen,
		output reg [1:0] a_dout, b_dout
	);
		reg [1:0] memory [0:3];

		initial begin
			memory[0] = 0;
			memory[1] = 1;
			memory[2] = 2;
			memory[3] = 3;
		end

		always @(posedge clk) begin
			if (a_wen)
				memory[a_addr] <= a_din;
			if (b_wen)
				memory[b_addr] <= b_din;
			a_dout <= memory[a_addr];
			b_dout <= memory[b_addr];
		end
	endmodule

In general you should only write to a memory from two always blocks if the writes are from two different clock domains.

I am not sure yet if I will create a work-around for this problem in Yosys. For this specific case the problem will go away anyway as soon as support for memory initialization is added.. I have to think a little more about it..

But given that Vivado's `synth_design` creates horrible results for small memories when the write port is implemented with two always blocks as shown above, and given the generic nature of the `BRAM_MEMORY_SP` module, it might be a good idea to rewrite it to use a single-always-block coding style anyways.

PS: Another problem is that the initialization values for FFs do not make it into the EDIF netlist at the moment. You have to convert it to an explicit reset path with a reset input pin, like you would in an ASIC flow (Sec. IV in [Appnote 011](http://www.clifford.at/yosys/files/yosys_appnote_010_verilog_to_blif.pdf) shows how it done). In short: Initialized data isn't really supported in Yosys 0.5 synthesis. (In my defense: It is well supported in the formal verification flow. ;) But I am aware that this is a very important feature and hope to have much better support for it in Yosys 0.6..

ferrandi, 2015-02-13:
Thanks for going in deep with the issue. It was and it is a pain to model BlockRam in Verilog for FPGA. In our framework we have different models for different FPGA vendors.
The example I've submitted is a customization of a more general case.
Anyway, the big problem I'd in modeling *BRAM_MEMORY_SP* is to use the byte-enabling feature. So, the solution I've found that pass all the regressions is to split reading from writing and to describe the byte enabling with the generate for. Altera for example goes on the same path you are suggesting but it is less parametrizable(e.g., I was not able to use generate for) and require SystemVerilog .
In more complex cases we have in our framework for Xilinx, the number of read and write ports could be different so I've other models where reading and writing are separated in different processes. This mainly comes from the fact that Xilinx supports multiple reads and up to two write ports.

>In general you should only write to a memory from two always blocks if the writes are from two different clock domains.
I do not have evidence of this from the Xilinx documentation. Maybe I miss something.

Anyway, thanks for looking in the issue.

cliffordwolf, 2015-02-13:
> So, the solution I've found that pass all the regressions is to split reading from writing and to describe the byte enabling with the generate for.

unrolling for loops in always blocks is part of synthesizable verilog. (sec. 7.7.6 of the 2005 Verilog RTL Synthesis Standard says they are supported under the condition that "Loop bounds shall be statically computable for a for loop.") So you can write everything in one always block by substituting generate-for with a regular for loop :

    integer i11;
    always @(posedge clock)
    begin
      if(HIGH_LATENCY==0)
      begin
        dout_a <= memory[memory_addr_a];
      end
      else
      begin
        dout_a_registered <= memory[memory_addr_a];
        dout_a <= dout_a_registered;
      end

      for (i11=0; i11<n_byte_on_databus; i11=i11+1)
      begin
          if(we_a[i11])
            memory[memory_addr_a][(i11+1)*8-1:i11*8] <= din_value_aggregated[(i11+1)*8-1:i11*8];
      end
      if (n_bytes > BRAM_BITSIZE/8)
      begin
        begin
          if(HIGH_LATENCY==0)
          begin
            dout_b <= memory[memory_addr_b];
          end
          else
          begin
            dout_b_registered <= memory[memory_addr_b];
            dout_b <= dout_b_registered;
          end
        end
        for (i11=0; i11<n_byte_on_databus; i11=i11+1)
          begin
            if(we_b[i11])
              memory[memory_addr_b][(i11+1)*8-1:i11*8] <= din_value_aggregated[(i11+1+n_byte_on_databus)*8-1:(i11+n_byte_on_databus)*8];
          end
      end
    end

>> In general you should only write to a memory from two always blocks if the writes are from two different clock domains.

> I do not have evidence of this from the Xilinx documentation. Maybe I miss something.

This is not Xilinx specific. The synthesis language spec does not define what should happen in those cases. So it is only supported for cases like clock domain crossing and is always a non-standard extension and should be assumed not to be supported unless stated differently in the documentation.

cliffordwolf, 2015-04-06:
Support for initialized (Xilinx) BRAMs has now been implemented in 8520b7fbe0b24dda47749aa870881b6b03480d4a. Please test.

I'm now closing this issue. Please open a new issue if you find a bug in the feature.

closed 2015-04-06 by cliffordwolf