YosysHQ/yosys #51: write_verilog -attr2comment outputs bad comments

jeremysalwen, 2015-02-13:
The output of 

    write_verilog -attr2comment

is not always a comment.  For example I get a line like

     /* src = "../folder/*srcfile.v */:3" */

Which has an extra  */ in the middle of it.

cliffordwolf, 2015-02-13:
That is interesting. How does it look like without -attr2comment? (all the option does is changing `(* .. *)` to `/* .. */`. The string inside is unchanged. I would guess the "*/" is actually the value or the `src` attribute. So the question is: where does it come from? What output do you get for the following command?

    ./yosys -qp 'synth; write_verilog -attr2comment' tests/simple/fiedler-cooley.v | grep src

I have now fixed printing of strings with `*/` in them for -attr2comment in commit 756b4064b29cf07bf0870f06dd538b34763f3cdd. This will now be printed as `*\/` which when interpreted with the verilog rules for strings will still be the string `*\`.

jeremysalwen, 2015-02-13:
The output of the command you posted is

```
/* src = "tests/simple/fiedler-cooley.v:3" */
  /* src = "tests/simple/fiedler-cooley.v:14" */
  /* src = "tests/simple/fiedler-cooley.v:14" */
  /* src = "tests/simple/fiedler-cooley.v:14" */
  /* src = "tests/simple/fiedler-cooley.v:14" */
  /* src = "tests/simple/fiedler-cooley.v:9" */
  /* src = "tests/simple/fiedler-cooley.v:9" */
  /* src = "tests/simple/fiedler-cooley.v:6" */
  /* src = "tests/simple/fiedler-cooley.v:12" */
  /* src = "tests/simple/fiedler-cooley.v:8" */
  /* src = "tests/simple/fiedler-cooley.v:5" */
  /* src = "tests/simple/fiedler-cooley.v:6" */
  /* src = "tests/simple/fiedler-cooley.v:9" */
  /* src = "tests/simple/fiedler-cooley.v:6" */
```

As for how that became part of the source string, I'm not sure.  I read in the file using 

      Pass::call(design,"read_verilog ../folder//srcfile.v");

Perhaps this is where the bug is coming from?  I figured the extra slash wouldn't change anything, as it found the file properly.

cliffordwolf, 2015-02-14:
Ah! The double slash is the problem. I can reproduce it with that:

    ./yosys -qp 'synth; write_verilog -attr2comment' tests/simple//fiedler-cooley.v | grep src

Looking into it now..

cliffordwolf, 2015-02-14:
Fixed in commit ef151b0b30c7b7253aceba6d184e7721a4eff0c0. Thanks for bringing this issue to my attention.

closed 2015-02-14 by cliffordwolf