YosysHQ/yosys #52: iterating wreduce enters infinite loop

jeremysalwen, 2015-02-15:
From my understanding of the documentation, wreduce is supposed to reduce the width of cells and wires.  It would expect that after one run (or perhaps a few), the width would be minimal, and wreduce would cease to change the design.

If you try the test code used in the documentation with wreduce

```
module test(input [3:0] a, b, c, output [7:0] y);
        assign y = a + b + c + 1;
    endmodule
```

running wreduce repeatedly does not converge, but rather it enters an infinite loop of "removing top bits" from wires.  I originally encountered this bug because it seemed wreduce was not removing unused bits in width 2 wires, or at least was introducing new wires with unused bits.

cliffordwolf, 2015-02-16:
It is not really doing anything here. The "removing top bits" does not actually remove the bits. It changes the wiring around and lets a later "clean" actually get rid of them. When you call "clean" between two calls to "wreduce" then the loop will stop. When you just call "wreduce" in a loop it will prepare this wires in a way that "clean" can get rid of them over and over again.. (It renames the wires, thus it looks like its a different wire in each iteration.)

I have now fixed this in commit 0748ef638d34bb5dd125271abc4fe051f95a790d. Now wreduce recognizes the already-prepared-for-deletion wires and does not touch them a second time.

closed 2015-02-16 by cliffordwolf

jeremysalwen, 2015-02-16:
Ah, I understand.  Perhaps a note about this could be added to the wreduce documentation/output that it needs to be run in conjunction with clean?

while you're at it there is a typo in the title of this page as well (is->if)
http://www.clifford.at/yosys/cmd_wreduce.html

I am under the impression that you are looking for bug reports/typos etc for even minor things, but if you are not, feel free to let me know so you can focus on higher priorities.

cliffordwolf, 2015-02-17:
> Perhaps a note about this could be added to the wreduce documentation 

This is a general theme in Yosys, not specific to the wreduce command. (The help message to opt_clean has something on it and I've now [added some additional explanations](https://github.com/cliffordwolf/yosys/commit/138547f41b5851abd7631074b1dbc3d6ef3fb4b3) to the CodingReadme file.)

> while you're at it there is a typo in the title of this page as well

fixed. (webpage will be updated when I release yosys 0.6)

> I am under the impression that you are looking for bug reports/typos etc for even minor things, ...

Yes. A steady flow of easy to fix bug reports is very welcome..  ;)

----

Let me just add a more detailed explanation of what was going on with wreduce (for the record):

Say you have a signal `a[5:0]` and only the lower 4 bits of it are used, so "wreduce" tries to "remove" the top two bits. I cannot remove a Wire object (or make it shorter, see the explanation in CodingReadme I mentioned above). So I instead create a new wire `$temp42[3:0]` and essentially add `assign $temp42[3:0] = a[3:0]` to the design. Finally I rename `$temp42` to `a` and `a` to `$temp42`. The next call to "clean" will remove the wire `$temp42`.

Before my bugfix, if you called "wreduce" a 2nd time it looked at the `$temp42[5:0]` wire and said "hay - I can remove the two top bits from it", create a  `$temp43[3:0]`, assign `$temp43[3:0] = $temp42[5:0]` and flip the names. Now after the bugfix "wreduce" understands that it is operating on wires that "clean" would throw away anyways and simply skips them.