YosysHQ/yosys #53: Failed to detect width for parameter with expression

gaomy3832, 2015-03-01:
Hi,

I try to parse some verilog code with parameterized module. The default value of some parameters are given by expressions rather than constant values. For example:

```
// multiply.v
`define WIDTH 16

module multiply (
    input [A_WIDTH-1:0] a,
    input [B_WIDTH-1:0] b,
    output [OUT_WIDTH-1:0] out
);
    parameter A_WIDTH = `WIDTH;
    parameter B_WIDTH = `WIDTH;
    parameter OUT_WIDTH = `WIDTH * 2;
    assign out = a * b;
endmodule
```

If I run
`./yosys -p "read_verilog multiply.v"`

I get the following error:

```
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `multiply.v' to AST representation.
Generating RTLIL representation for module `\multiply'.
ERROR: Failed to detect width for parameter \OUT_WIDTH at multiply.v:6!
```

Would you consider fixing this bug, or do you have any suggestion to bypass this issue?

Thanks!

cliffordwolf, 2015-03-01:
You can work around that problem by making sure the parameter is defined before it is being used. You can do that by either going all-in with the Verilog-2001 syntax for port and parameter declarations:

    module test2 #( 
        parameter WIDTH = 16 * 2
    ) (
        output [WIDTH-1:0] out
    );
    endmodule

Or by falling back to the Verilog-95 syntax:

    module test3 (out);
        parameter WIDTH = 16 * 2;
        output [WIDTH-1:0] out;
    endmodule

I'll see if I can fix this and post another comment then.

cliffordwolf, 2015-03-01:
Fixed in commit 422794c5840784cab11930cfa305ed9351bb1a98.

closed 2015-03-01 by cliffordwolf

gaomy3832, 2015-03-02:
Thanks for the quick reply and fix! That's really helpful!