YosysHQ/yosys #54: error in 'make test'

bhaskarjk, 2015-03-20:
Hi Clifford,

I am trying to install yosys (latest git clone), and am getting the following error while running 'make test'. Similar errors seen for test/share & test/fsm. 

I'm on Centos 6.4 with python version 2.6.6. Are there any python version dependencies? Couldn't find any mention on your website/README, hence the question.

--- begin error message ---
cd tests/realmath && bash run-test.sh
generating tests..
  File "generate.py", line 43
    with file('temp/uut_%05d.v' % idx, 'w') as f, redirect_stdout(f):
                                                ^
SyntaxError: invalid syntax
make: *** [test] Error 1
--- end error message ---

Thanks & regards,
Bhaskar

cliffordwolf, 2015-03-20:
I have no python 2.6.6 for testing but I think I have fixed that now int commit 604c097f9818a5b29fa06223b88930630fafca5e. Please give it a try and let me know if that solved the issue for you.

bhaskarjk, 2015-03-20:
Hi Clifford,

Thanks for the prompt reply. I will try a git pull in a bit and confirm.

Regards,
Bhaskar Karmakar
S3Craft Technologies
Bangalore, India

-original message-
Subject: Re: [yosys] error in 'make test' (#54)

https://github.com/cliffordwolf/yosys/issues/54#issuecomment-83946735

bhaskarjk, 2015-03-20:
Clifford,

The 'make test' errors have been fixed. Thanks.

I see 2 warnings as below, maybe you know of them already.

cd tests/various && bash run-test.sh
Running constmsk_test.ys..
Warning: Range select out of bounds on signal `\tmp' at constmsk_testmap.v:45: Setting 1 MSB bits to undef.

cd tests/sat && bash run-test.sh
Running initval.ys..
Warning: ignoring initial value on non-register: \bar [1:0]

Regards,
Bhaskar

-----Original Message-----
From: Bhaskar Karmakar [mailto:bhaskar@s3craft.com] 
Sent: 20 March 2015 14:06
To: cliffordwolf/yosys
Cc: bhaskar@s3craft.com
Subject: RE: [yosys] error in 'make test' (#54)

Hi Clifford,

Thanks for the prompt reply. I will try a git pull in a bit and confirm.

Regards,
Bhaskar Karmakar
S3Craft Technologies
Bangalore, India

-original message-
Subject: Re: [yosys] error in 'make test' (#54)

https://github.com/cliffordwolf/yosys/issues/54#issuecomment-83946735

cliffordwolf, 2015-03-20:
Thanks for verifying the fix. Those two warning are ok. This test cases deliberately test if yosys does the right thing in the situations that trigger those warnings..

closed 2015-03-20 by cliffordwolf