YosysHQ/yosys #56: Extract parameterized subcircuit

gaomy3832, 2015-04-07:
Hi,

I am trying to use the extract command to extract a subcircuit in my design. However, the reference circuit is parameterized, and the default parameters do not match the ones used in the top design. In such a case, yosys fails to detect the subcircuit. I tried to change the default parameters in the reference design to match the top design, then it worked. I also tried -ignore_parameters option, it did not work. I would like a general solution which can handle different parameter values. Is there a way to do this?

Thanks!

cliffordwolf, 2015-04-07:
If I understand you correctly, your needle circuit itself is parameterized. The -ignore_parameters option tells the extract command to ignore parameters when matching cells. The parameters of the needle module itself are applied by the verilog front-end when reading the module and are of no consequence for the extract command.

So different parameter sets for your needle circuit basically mean different needle circuits. You have to create those different needle modules and pass them to "extract" as separate circuits.

I have now implemented a "chparam" command in 329b841aac58028de51ecd65262f0dda546abf5c that simplifies creating different needle modules in such a situation:

    read_verilog <<EOT
      module test #(parameter SZ = 4) (input [SZ-1:0] A, B, output [SZ-1:0] Y);
        assign Y = A + B;
      endmodule
    EOT
    
    copy test test6
    copy test test8
    
    chparam -set SZ 6 test6
    chparam -set SZ 8 test8

    delete test
    write_ilang mycells.il

So now you can use something like "extract -map mycells.il" to extract test6 and test8 cells from a circuit.

closed 2015-04-07 by cliffordwolf

gaomy3832, 2015-04-07:
Thanks for the quick response and solution. I will try it out.

Also, is there a way to get the parameter set of the haystack circuit? I just want to automate the procedure a bit further. If not it is also fine.

cliffordwolf, 2015-04-07:
I have now implemented "chparam -list" in commit 1f33b2a490cf44ec24de5e085ff2957a680519fb. 

Unfortunately it is not possible to recover the parameter values after the verilog front-end has been executed (they are not preserved in a way that would be easily accessible). But this will at least give print the parameter names. (Not sure if that is helpful in your application, but I imagine there are situations where this could be useful and it was a simple feature to add.)