YosysHQ/yosys #57: Synposys gatelibs?

jeremysalwen, 2015-04-07:
Synopsys tools output cells like "INVX1", etc, which yosys does not recognize as logical operations.  What would be the best way to get yosys to understand designs like this?  Is there a publicly available gatelib for these cells?

cliffordwolf, 2015-04-07:
Since I'm not a synopsys user myself I can't say if there is a complete publicly available library of those cells. But there are some liberty files that implement at least what seems to be a subset. For example: http://vlsiarch.ecen.okstate.edu/flows/MOSIS_SCMOS/latest/cadence/lib/tsmc025/signalstorm/osu025_stdcells.lib

For example with that liberty file, and a structural netlist synth.v, you can do the following to replace the library cells in the design with their definitions from the cell library:

    # read structural netlist
    read_verilog synth.v

    # replace library cells
    read_liberty -setattr libcells osu025_stdcells.lib
    flatten A:libcells %C
    delete A:libcells

(This requires git head. I've just implemented the select operator %C in commit 8eadd8fb18eb0a0707dd5f3262adf2e811a5bc69 because I realized that this functionality was missing when I started to answer your question.)

closed 2015-04-07 by cliffordwolf

jeremysalwen, 2015-04-08:
Could you explain how this is different than using techmap?

cliffordwolf, 2015-04-08:
Techmap and flatten actually share the same codebase. So its not different at all. But techmap can only read verilog and ilang files at the moment, so it was not the first thing I thought of and then I decided I wanted to implement the %C (and %M) select operators..

You are right: using techmap would actually be much better. (And works with yosys-0.5 just fine.)

So you would either convert the liberty file to ilang first:

    read_liberty osu025_stdcells.lib
    write_ilang libcells.il
    design -reset

    read_verilog synth.v
    techmap -map libcells.il

Or (a bit cleaner imo) use separate designs in memory:

    read_liberty osu025_stdcells.lib
    design -stash libcells

    read_verilog synth.v
    techmap -map %libcells

So that's proof that just because I wrote the software doesn't mean that I always intuitively know how to use it best.. :)