YosysHQ/yosys #65: Verilog generate with decrement downto 0 causes yosys to hang

m-temp, 2015-05-29:
It seams like there's a problem with the verilog "generate" statement and a decrement downto 0.

Have a look at the following example:

```
module test(
        alu_data_d_in,
	alu_data_d_out
    );

    input [7:0]alu_data_d_in;
    output[7:0]alu_data_d_out;
    wire [7:0]swap_out;
    genvar i;

    generate
        for ( i = 7 ; ( i >= 4 ) ; i = ( i - 1 ) )
        begin : swap_h
            assign swap_out[i] = alu_data_d_in[( ( i - 4 ) )];
        end
    endgenerate
     generate
         //for ( i = 0 ; ( i <4 ) ; i = ( i + 1 ) )  //OK
         for ( i = 3 ; ( i >=0 ) ; i = ( i - 1 ) ) //FAIL
         begin : swap_l
             assign swap_out[i] = alu_data_d_in[(i+4 )];
         end
     endgenerate

assign alu_data_d_out = swap_out;

endmodule 
```

If I use " for ( i = 3 ; ( i >=0 ) ; i = ( i - 1 ) )" instead of "for ( i = 0 ; ( i <4 ) ; i = ( i + 1 ) )" yosys hangs (100% load on one cpu core) without error message and starts consuming more and more memory.
It seams like there is a problem with the 0, because " for ( i = 7 ; ( i >= 4 ) ; i = ( i - 1 ) )" does work. 
Also " for ( i = 3 ; ( i =0 ) ; i = ( i - 1 ) )" runs. (However the result is wrong...)

udif, 2015-05-29:
Just a guess - Maybe yosys treats genvars as unsigned?

cliffordwolf, 2015-05-29:
I've now fixed the signdness of genvar expressions in commit 99b8746d277522cba2516747666645adb8f81a68. Thanks for reporting this issue.

closed 2015-05-29 by cliffordwolf