YosysHQ/yosys #79: Segfault running synthesis on Raptor64 CPU core

kallisti5, 2015-09-06:
verilog :) $ gdb yosys -p 'synth_ice40 -top Raptor64 -blif example.blif' Raptor64*.v

{{{
(gdb) run -p 'synth_ice40 -top Raptor64 -blif example.blif' Raptor64*.v
Starting program: /usr/local/bin/yosys -p 'synth_ice40 -top Raptor64 -blif example.blif' Raptor64*.v
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2015  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.5+249 (git sha1 8d6d5c3, clang 3.6.2 -fPIC -Os)

-- Parsing `Raptor64_addsub.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `Raptor64_addsub.v' to AST representation.
Generating RTLIL representation for module `\Raptor64_addsub'.
Note: Assuming pure combinatorial block at Raptor64_addsub.v:45 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Successfully finished Verilog frontend.

-- Parsing `Raptor64_bitfield.v' using frontend `verilog' --

2. Executing Verilog-2005 frontend.
Parsing Verilog input from `Raptor64_bitfield.v' to AST representation.
Generating RTLIL representation for module `\Raptor64_bitfield'.
Note: Assuming pure combinatorial block at Raptor64_bitfield.v:54 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Note: Assuming pure combinatorial block at Raptor64_bitfield.v:58 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Successfully finished Verilog frontend.

-- Parsing `Raptor64_BranchHistory.v' using frontend `verilog' --

3. Executing Verilog-2005 frontend.
Parsing Verilog input from `Raptor64_BranchHistory.v' to AST representation.
Generating RTLIL representation for module `\Raptor64_BranchHistory'.
Warning: Blocking assignment to memory in line Raptor64_BranchHistory.v:46 is handled like a non-blocking assignment.
Note: Assuming pure combinatorial block at Raptor64_BranchHistory.v:64 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Successfully finished Verilog frontend.

-- Parsing `Raptor64_BypassMux.v' using frontend `verilog' --

4. Executing Verilog-2005 frontend.
Parsing Verilog input from `Raptor64_BypassMux.v' to AST representation.
Generating RTLIL representation for module `\Raptor64_BypassMux'.
Note: Assuming pure combinatorial block at Raptor64_BypassMux.v:41 in
[... 266 lines cut ...]
    at frontends/ast/ast.cc:904
#6  0x000000000055edcb in Yosys::AST::process (design=0xb8adf0, ast=<optimized out>, 
    dump_ast1=<optimized out>, dump_ast2=<optimized out>, dump_vlog=64, nolatches=72, 
    nomeminit=<optimized out>, nomem2reg=<optimized out>, mem2reg=<optimized out>, lib=<optimized out>, 
    noopt=<optimized out>, icells=<optimized out>, ignore_redef=<optimized out>, defer=<optimized out>, 
    autowire=<optimized out>) at frontends/ast/ast.cc:1014
#7  0x0000000000559743 in Yosys::VerilogFrontend::execute (this=<optimized out>, f=<optimized out>, 
    filename=..., args=..., design=0xb8adf0) at frontends/verilog/verilog_frontend.cc:316
#8  0x00000000004b8fe4 in Yosys::Frontend::execute (this=0xb6d2f0 <Yosys::VerilogFrontend>, 
    args=std::vector of length 2, capacity 2 = {...}, design=0xb8adf0) at kernel/register.cc:311
#9  0x00000000004b9b97 in Yosys::Frontend::frontend_call (design=0xb8adf0, f=<optimized out>, 
    filename=..., args=std::vector of length 2, capacity 2 = {...}) at kernel/register.cc:422
#10 0x00000000004b998a in Yosys::Frontend::frontend_call (design=0xb8adf0, f=0x0, filename=..., 
    command=...) at kernel/register.cc:400
#11 0x0000000000501f01 in Yosys::run_frontend (filename=..., command=..., 
    backend_command=<optimized out>, from_to_label=<optimized out>, design=<optimized out>)
    at kernel/yosys.cc:834
#12 0x00000000004b5daf in main (argc=29, argv=0x7fffffffe508) at kernel/driver.cc:347
(gdb) quit
A debugging session is active.

	Inferior 1 [process 8416] will be killed.

Quit anyway? (y or n) y
}}}

cliffordwolf, 2015-09-07:
I cannot reproduce this. So questions:

* What Linux distribution and version are you running on? Anything I should know if I want to install it in a VM to best match your setup?

* Do you also get a segfault if you simply run `yosys Raptor64mc.v`?

* Which version of the Raptor64 core are you using? I just downloaded the "latest" .tar.gz from opencores: `md5sum Raptor64mc.v` -> `533532e5722ff1eb15d9d3fb5b20add6`

cliffordwolf, 2015-09-18:
I'm now closing this issue as I cannot reproduce the problem and the further information I requested in my previous comment (11 days ago) was not provided.

Please feel free to provide further information on how to reproduce this problem (Linux distribution and version, etc.) and reopen this issue at any time.

closed 2015-09-18 by cliffordwolf