YosysHQ/yosys #81: Segfault found with fuzzer

jeremysalwen, 2015-09-22:
I am running a fuzzer on yosys.  This is the first segfault I have found.  I believe it is caused by the "1'b_ "

```

module demo_001(y1, y2, y3, y4);
	output [7:0] y1, y2, y3, y4;

	localparam [7:0] p1 = 123.45;
	localparam real p2 = 123.45;
	localparam real p3 = 123;
	localparam p4 = 123.45;

	assign y1 = p1 + 0.2;
	assign y2 = p2 + 0.2;
	assign y3 = p3 + 0.2;
	assign y4 = p4 + 0.2;
endmodule

module demo_002(y0, y1, y2, y3);
	output [63:0] y0, y1, y2, y3;

	assign y0 = 1'b_ >= (-1 * -1.17);
	assign y1 = 1 ?  1 ?  -1 : 'd0 : 0.0;
	assign y2 = 1 ? -1 :   1 ? 'd0 : 0.0;
	assign y3 = 1 ? -1 : 'd0;
endmodule

```
```
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `/tmp/lol.v' to AST representation.

Program received signal SIGSEGV, Segmentation fault.
Yosys::my_strtobin (data=..., str=<optimized out>, len_in_bits=1, base=<optimized out>, 
    case_type=<optimized out>) at frontends/verilog/const2ast.cc:125
125		RTLIL::State msb = data.back();
(gdb) back
#0  Yosys::my_strtobin (data=..., str=<optimized out>, len_in_bits=1, base=<optimized out>, 
    case_type=<optimized out>) at frontends/verilog/const2ast.cc:125
#1  0x000000000095e456 in Yosys::VERILOG_FRONTEND::const2ast (code=..., case_type=0 '\000', 
    warn_z=<optimized out>) at frontends/verilog/const2ast.cc:213
#2  0x000000000095d40f in Yosys::VERILOG_FRONTEND::const2ast (code=..., case_type=0 '\000', 
    warn_z=<optimized out>) at frontends/verilog/const2ast.cc:153
#3  0x0000000000911f6d in frontend_verilog_yyparse () at frontends/verilog/verilog_parser.y:1303
#4  0x000000000095a696 in Yosys::VerilogFrontend::execute (this=<optimized out>, f=<optimized out>, 
    filename=..., args=..., design=0x1f8cec0) at frontends/verilog/verilog_frontend.cc:313
#5  0x000000000049fca2 in Yosys::Frontend::execute (this=0x1f6f3a0 <Yosys::VerilogFrontend>, args=warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.: 

std::vector of length 64, capacity 64, design=0x1f8cec0) at kernel/register.cc:311
#6  0x00000000004a681d in Yosys::Frontend::frontend_call (design=0x1f8cec0, f=<optimized out>, 
    filename=..., args=warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.: 
std::vector of length 64, capacity 64) at kernel/register.cc:422
#7  0x00000000004a4b20 in Yosys::Frontend::frontend_call (design=0x1f8cec0, f=0x0, filename=..., 
    command=...) at kernel/register.cc:400
#8  0x000000000070daeb in Yosys::run_frontend (filename=..., command=..., 
    backend_command=<optimized out>, from_to_label=<optimized out>, design=<optimized out>)
    at kernel/yosys.cc:834
#9  0x000000000048c2b3 in main (argc=2, argv=0x7fffffffe378) at kernel/driver.cc:347
```

cliffordwolf, 2015-09-22:
Yes, the problem was the `1'b_`. This should be a syntax error, but it was easier to fix it so that this is the same as `1'b_0`, which I have now done in commit 4b8200eb4943dcbdb. 

Reduced test case:

    module demo(output [63:0] y);
      assign y = 1'b_;
    endmodule

Apparently there is a similar bug in Icarus Verilog:

    $ iverilog test.v 
    ivl: verinum.cc:364: verinum::V verinum::get(unsigned int) const: Assertion `idx < nbits_' failed.
    sh: line 1:  2885 Done                    /usr/local/lib/ivl/ivlpp -L -F"/tmp/ivrlg25884e2fb" -f"/tmp/ivrlg5884e2fb" -p"/tmp/ivrli5884e2fb"
          2886 Aborted                 | /usr/local/lib/ivl/ivl -C"/tmp/ivrlh5884e2fb" -C"/usr/local/lib/ivl/vvp.conf" -- -

Do you want to [report that](https://github.com/steveicarus/iverilog/issues) or should I?

Thanks for the bug report. Please keep fuzzing..

closed 2015-09-22 by cliffordwolf

jeremysalwen, 2015-09-22:
You should report it, I'm not familiar with icarus verilog.

hailinzeng, 2015-10-10:
Hi, @jeremysalwen , do you have any plan to open source your AFL dictionary for verilog?

jeremysalwen, 2015-10-10:
@hailinzeng  Yes, certainly.  I just took the list of keywords from here: https://www.csee.umbc.edu/portal/help/VHDL/verilog/reserved.html  and combined it with a list of all string constants from the yosys source code that began with backslash.  http://pastebin.com/zzLDvLak