YosysHQ/yosys #83: Segfault in simplify

jeremysalwen, 2015-09-22:
Detected using AFL.  I think this is a stack overflow

```
module a(b);
input b;
reg c;parameter
signed b=b;
endmodule

```

```
-- Parsing `/tmp/lol.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `/tmp/lol.v' to AST representation.
Generating RTLIL representation for module `\a'.

Program received signal SIGSEGV, Segmentation fault.
0x00000000009857cc in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=true, at_zero=false, in_lvalue=false, stage=1, width_hint=-1, 
    sign_hint=<error reading variable: Cannot access memory at address 0x0>, in_param=<error reading variable: Cannot access memory at address 0x7fffff7feb90>) at frontends/ast/simplify.cc:51
51	{
(gdb) back
#0  0x00000000009857cc in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=true, at_zero=false, in_lvalue=false, stage=1, width_hint=-1, 
    sign_hint=<error reading variable: Cannot access memory at address 0x0>, in_param=<error reading variable: Cannot access memory at address 0x7fffff7feb90>) at frontends/ast/simplify.cc:51
#1  0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff802b18: -1, sign_hint=@0x7fffff802b17: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#2  0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)
    at frontends/ast/simplify.cc:309
#3  0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff804c78: -1, sign_hint=@0x7fffff804c77: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#4  0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)
    at frontends/ast/simplify.cc:309
#5  0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff806dd8: -1, sign_hint=@0x7fffff806dd7: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#6  0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)
    at frontends/ast/simplify.cc:309
#7  0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff808f38: -1, sign_hint=@0x7fffff808f37: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#8  0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)
    at frontends/ast/simplify.cc:309
#9  0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff80b098: -1, sign_hint=@0x7fffff80b097: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#10 0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)
    at frontends/ast/simplify.cc:309
#11 0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff80d1f8: -1, sign_hint=@0x7fffff80d1f7: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#12 0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)
    at frontends/ast/simplify.cc:309
#13 0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff80f358: -1, sign_hint=@0x7fffff80f357: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#14 0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)
    at frontends/ast/simplify.cc:309
#15 0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff8114b8: -1, sign_hint=@0x7fffff8114b7: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#16 0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)
    at frontends/ast/simplify.cc:309
#17 0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff813618: -1, sign_hint=@0x7fffff813617: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#18 0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)
    at frontends/ast/simplify.cc:309
#19 0x00000000009fec80 in Yosys::AST::AstNode::detectSignWidthWorker (this=0x1f95db0, width_hint=@0x7fffff815778: -1, sign_hint=@0x7fffff815777: true, found_real=<optimized out>)
    at frontends/ast/genrtlil.cc:571
#20 0x0000000000991887 in Yosys::AST::AstNode::simplify (this=0x1f95eb0, const_fold=<optimized out>, at_zero=false, in_lvalue=false, stage=1, 
    width_hint=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, sign_hint=<error reading variable: Cannot access memory at address 0x1>, in_param=<optimized out>)

```

cliffordwolf, 2015-09-22:
Interesting. It should have printed

    Warning: Deep recursion in AST simplifier.
    Does this design contain insanely long expressions?

before segfaulting. And it does here on my machine.

Seems like it crashed for you before it reached 1000 recursion levels. Did you disable compiler optimizations or did anything else that could have caused the size of the simplifier stack frame to increase?

Here is what happens:

    module test;
      parameter x=x;
    endmodule

The simplifier tries to substitute the value of the constant `x`, so it looks `x` up sees that its a parameter, tries to evaluate its constant value and so on.

It is not so trivial to detect all possible cases of something like that. In case of recursive function calls for example (done via exactly the same mechanism) it is impossible to tell if a recursion will terminate, because it would be equivalent to solving the [halting problem](https://en.wikipedia.org/wiki/Halting_problem).

So even though this specific case would be easy to fix, I won't bother because its just a trivial example of an inherently unfixable problem.

It would be interesting to find out why you don't see the `Warning: Deep recursion in AST simplifier.` message. Maybe you could adjust the threshold `recursion_counter++ == 1000` in `simplify.cc` to something like `recursion_counter++ == 100`. Then you can simply ignore all test cases that produce this warning message..

> I think this is a stack overflow

Well, yes, this is a stack overflow in the sense that the stack runs out of space, not in the sense that management data on the stack frame is overwritten. So its essentially an out of memory error.

closed 2015-09-22 by cliffordwolf

jeremysalwen, 2015-09-23:
If I don't run it through gdb it produces the warning message you mentioned.

I was under the impression that any segfault is a security risk.  Googling for information about infinite recursion as a security risk yields ambiguous information.  It seems like it can be on unusual systems.

For fuzzing purposes I just made it abort instead of warning.
http://stackoverflow.com/questions/4104057/is-a-stack-overflow-a-security-hole
http://en.0day.today/exploit/22695

cliffordwolf, 2015-09-23:
> I was under the impression that any segfault is a security risk.

Even if it was.. Yosys has DPI-C support, so just consider the following:

    $ cat test.v
    module test;
      import "DPI-C" function system = integer system(integer);
      integer foo = system(23);
    endmodule

    $ strace -f -e execve ./yosys test.v
    [...]
    1. Executing Verilog-2005 frontend.
    Parsing Verilog input from `test.v' to AST representation.
    Generating RTLIL representation for module `\test'.
    Calling DPI function `system' and returning `integer':
      arg 0 (integer): 23
    Process 25595 attached
    [pid 25595] execve("/bin/sh", ["sh", "-c", 0x17], [/* 99 vars */]) = -1 EFAULT (Bad address)
    [pid 25595] +++ exited with 127 +++
    --- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=25595, si_status=127, si_utime=0, si_stime=0} ---
      return integer: 32512
    Successfully finished Verilog frontend.

jeremysalwen, 2015-09-23:
You don't need to do call the 'plugin' pass first to use DPI-C functions?  It does surprise me that parsing a verilog file would allow arbitrary code execution.

I am using yosys for the frontend to my zero-knowledge proof protocol, which is supposed to be secure against malicious inputs.  I'm not too concerned about the infinite recursion, but is there a way to run yosys that will disable the DPI-C loophole?

cliffordwolf, 2015-09-23:
> You don't need to do call the 'plugin' pass first to use DPI-C functions?

No. A common use-case for DPI-C is actually just calling `math.h` functions that have no equivalent verilog system function.

> is there a way to run yosys that will disable the DPI-C loophole?

I have now added a `-nodpi` option to `read_verilog`: e2e092b14406fdf0ad71bc2821fe2c880d927484