YosysHQ/yosys #84: Possible bug with memory_bram

gaomy3832, 2015-09-23:
I find a bug with memory_bram. Not sure if it is a real bug or I just misused something, but here it is:

```
module ram_block (addr, ce, q, clk);

input clk;
input [9:0] addr;
input ce;
output reg [7:0] q;

reg [7:0] ram[1023:0];

always @(posedge clk)
begin
    if (ce)
    begin
        q <= ram[addr];
    end
end

endmodule
```

If I just run `./yosys -p "read_verilog ram_block.v; synth_xilinx -top ram_block"`, it will segfault during memory_bram pass:

```
...
...

2.7. Executing MEMORY_BRAM pass (mapping $mem cells to block memories).
Processing ram_block.ram:
  Properties: ports=1 bits=8192 rports=1 wports=0 dbits=8 abits=10 words=1024
  Checking rule #1 for bram type $__XILINX_RAMB36_SDP (variant 1):
    Bram geometry: abits=9 dbits=72 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__XILINX_RAMB36_SDP: awaste=0 dwaste=64 bwaste=32768 waste=32768 efficiency=11
    Rule #1 for bram type $__XILINX_RAMB36_SDP (variant 1) accepted.
    Mapping to bram type $__XILINX_RAMB36_SDP (variant 1):
      Shuffle bit order to accommodate enable buckets of size 9..
      Results of bit order shuffling: 0 1 2 3 4 5 6 7 -1
Segmentation fault
```

cliffordwolf, 2015-09-24:
Thanks for bringing this issue to my attention. Fixed in commit 3501f8e3643bfb6cd13f8d6e1acb03fa6672fd27.

closed 2015-09-24 by cliffordwolf

gaomy3832, 2015-09-24:
Thanks, the patch fixes the segfault problem.

However, for such a verilog declaration for memory, Yosys recognizes it as an async memory, and therefore fails to map it to a clocked BRAM in xilinx/brams.txt (it does map to drams which is non-clocked).

I tracked down a little bit and found that there is a MUX between the $mem and $dff, due to the ce signal. The ce can be actually viewed as an enable signal to the memory. Is it possible to add some features for memory_dff pass to also work with this style?

Just FYI, this code style is generated from Xilinx Vivado High Level Synthesis tool, so it is not easy for me to just change the verilog code.

cliffordwolf, 2015-09-25:
Adding read-enable inputs to the Yosys memory model is on my todo list for quite a while now. (Adding support to `memory_dff` is only a small fraction of that.) Might as well do it now: BRAMs with read-enables are now supported starting with commit 924d9d6e86a5e9a2294479345daac1c03d78008a.

gaomy3832, 2015-09-25:
Thanks so much for such a prompt response. It works nicely on my current design. I will reach back to you if any bug is found.

gaomy3832, 2015-09-30:
I have another case for the read-enable input. Here is a transparent read/write port RAM

https://github.com/gaomy3832/yosys_tests/blob/master/rw_port_ram.v

I ran `./yosys -p "read_verilog rw_port_ram.v; synth_xilinx -top rw_port_ram -run :dram; stat"`. The `$mem` cell can't be mapped to BRAM, even when I changed the BRAM rule file to have configurable transp read ports.

cliffordwolf, 2015-10-01:
Right now Yosys only recognizes the "dff-on-read-address" coding style for transparent read ports. I think the following module does what you want:

    module rw_port_ram (addr0, ce0, we0, d0, q0, clk);
    
    input clk;
    input [11:0] addr0;
    input ce0;
    input we0;
    input [31:0] d0;
    output [31:0] q0;
    
    reg [31:0] ram[4095:0];
    reg [11:0] addr0_q;
    
    assign q0 = ram[addr0_q];
    
    always @(posedge clk)
    begin
        if (ce0)
        begin
            if (we0)
                ram[addr0] <= d0;
            addr0_q <= addr0;
        end
    end
    
    endmodule