YosysHQ/yosys #87: Performance issues with memory initialization

gaomy3832, 2015-09-30:
I have two performance issues with the memory initialization. I am not sure if there are good and easy solutions, or even it is worth it to improve (while I do think these are moderate size applications), but just want to report these issues to motivate some discussion.

First, the verilog frontend runs really slow if there is an `initial` block with a moderate-sized `$readmemh` function (e.g., 4096 * 32). I attach an example here

https://github.com/gaomy3832/yosys_tests/blob/master/rom.v
https://github.com/gaomy3832/yosys_tests/blob/master/rom_ram.dat

If I run `./yosys -p "read_verilog rom.v;"`, it takes about 15 min.

```
CPU: user 875.76s system 0.06s, MEM: 110.15 MB total, 86.50 MB resident
Yosys 0.5+332 (git sha1 ddcfc99, clang 3.4-1ubuntu3 -fPIC -Os)
Time spent: 100% 2x read_verilog (875 sec)
```

Another issue is a little complex. What I want to do is to techmap the BRAM into my custom 1-bit cells (i.e., split the columns). For example, mapping a 64 * 8 BRAM into eight 64 * 1 BRAMs. If I want to preserve the initial values, I need to extract each column from the `INIT` parameter, which I assume is stored in row major format. So I try to use a constant function to do the transpose, see the example code here

https://github.com/gaomy3832/yosys_tests/blob/master/transpose.v

If I run `./yosys -p "read_verilog transpose.v"`, it doesn't finish after 30 min (I have to kill it).

I am not sure if using constant function is the best solution, so please let me know if yosys supports a better way to do the work.

A final note for my environment, I ran these tests on a server-class CPU (Xeon E5-2630) so my system should have enough processing power. I've also disabled tcl, abc, and readline when compiling, but I don't think it has any significant effect.

cliffordwolf, 2015-09-30:
Easy things first: I have now fixed the `$readmem` performance issue in commit 9caeadf79713534a1fd5f51fb458af245c34dad9.

Regarding block ram mapping:

You might want to consider using the `memory_bram` pass to map the memory to the type of block rams you have, and then use techmap to implement a 1:1 mapping of the cells created by `memory_bram` to the cells you have in your architecture. For example with your `rom.v` test case:

First you need to create a rules file for `memory_bram`:

    $ cat my_brams.txt 
    bram MY_64x1_BRAM
      init 1
      abits 6
      dbits 1
      groups 2
      ports  1 1
      wrmode 0 1
      enable 0 1
      transp 0 0
      clocks 0 1
      clkpol 0 1
    endbram
    
    match MY_64x1_BRAM
      min efficiency 10
      make_outreg
    endmatch

Then you can do something like the following in synthesis:

    read_verilog rom.v
    synth -run :fine
    memory_bram -rules my_brams.txt

After that you only need a simple techmap rule that maps MY_64x1_BRAM cells to the target memory cell type.

Please let me know if this is an OK solution to you so I can either close the issue or think about a different solution..

gaomy3832, 2015-09-30:
The first fix works quite well. Thanks!

For the second one, your suggestion works if all BRAMs are detected from verilog declaration like `reg [7:0] ram [1023:0]`. However, I also have a few blackbox RAM modules in the design that are not handled by `memory_bram`, but I do want to techmap them into the BRAM cells in the architecture. In that case I have to deal with the INIT value translation.

This is not an urgent issue as my goal is to evaluate the resource usage. Just ignoring the INIT value will not change the results. So if it is tricky, we can stash this issue and revisit it when really necessary.

Thanks again!

cliffordwolf, 2015-10-01:
I have now fixed a performance bug related to assigning to selections from bit vectors in constant functions in commit e51dcc83d087d2fc91c39460b3e8c5dcc683a4da. Now your `init_by_column` functions runs in a few seconds. Still pretty slow, but at least "reasonably slow".. ;)

You can further speed it up by using the following variant of your function that reduces the number of large copies a little bit (assigning a single bit in a vector like `init_by_row` still creates a complete copy each time):

    function [36863:0] init_by_column;
        input [36863:0] init_by_row;
        reg [511:0] single_column;
        integer idx, jdx;
        begin
            for (idx = 0; idx < 72; idx = idx + 1) begin
                for (jdx = 0; jdx < 512; jdx = jdx + 1) begin
                    single_column[jdx] = init_by_row[jdx * 72 + idx];
                end
                init_by_column[idx * 512 +: 512] = single_column;
            end
        end
    endfunction

closed 2015-10-01 by cliffordwolf

gaomy3832, 2015-10-01:
This fix is good enough in my case. And your note about large copies is very useful, too. Thanks!