YosysHQ/yosys #88: Parameter outside module yields syntax error

troelsfr, 2015-10-06:
Hey,

I get following error:

 ```/----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
(....)
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.5+336 (git sha1 e51dcc8, clang 3.4-1ubuntu3 -fPIC -Os)

yosys> read_verilog blah.v 
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `blah.v' to AST representation.
ERROR: Parser error in line blah.v:1: syntax error
tfr@tfr-laptop:~/Documents/TestArea/Yosys$ cat blah.v 
parameter X = 2;```

This is just a minimal example of the problem. The problem occurs when I define a parameter outside of a Verilog module. If inside the module, everything is fine:

```yosys> read_verilog blah.v
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `blah.v' to AST representation.
Generating RTLIL representation for module `\top'.
Successfully finished Verilog frontend.

yosys> exit

End of script. Logfile hash: bd98451c1e
CPU: user 0.00s system 0.00s, MEM: 43.50 MB total, 5.19 MB resident
Yosys 0.5+336 (git sha1 e51dcc8, clang 3.4-1ubuntu3 -fPIC -Os)
Time spent: 100% 2x read_verilog (0 sec)
tfr@tfr-laptop:~/Documents/TestArea/Yosys$ cat blah.v
module top(A);
   input[7:0] A;   
   parameter X = 2;
endmodule // top
```

cliffordwolf, 2015-10-07:
Verilog parameters must be contained within a module. I had a quick look into the SystemVerilog spec and it does not look like they allow parameters outside of modules either...

But maybe I'm wrong.. What makes you believe that verilog parameters outside a module are allowed in Verilog? What would the intended behavior of that be?

Closing the issue now. I'll reopen it if it turns out there is more to it..

closed 2015-10-07 by cliffordwolf

troelsfr, 2015-10-07:
Well, both Vivado and Verilator supports it, so I just naively assumes that it is valid. Of course, this does not mean that it is in the standard.

cliffordwolf, 2015-10-07:
> both Vivado and Verilator supports it

interesting. Icarus Verilog, GPLCver, Veriwell, Verific, and Synplify Pro produce errors for this. Riviera-PRO and VCS seem to accept parameters outside of modules.

Looking at the IEEE 1800-2012 Std. document again it looks like `parameter` declarations (as well as `localparam` declarations) are package items and package items can be used in the global context. So I have now added support for that in commit ba4cce9f19cf7a8eab78b52ca1aaa4d040eb8e7c.

Thanks for bringing this to my attention.

troelsfr, 2015-10-07:
Excellent - then I don't need to change my code right away, thanks!

From what you say, it sounds like it is left undefined in the standard, so probably I should get rid of global parameter definitions.

Thanks again.

cliffordwolf, 2015-10-07:
> it sounds like it is left undefined in the standard

Well, it's definitely not part of Verilog 2005. But it is part of SystemVerilog 2012 and seems to be well-defined.

However, I wonder if you are using `parameter` when in fact you'd only need `localparam`. (`parameter` can be overwritten at module instantiation, `localparam` is just a name for a constant value.)

troelsfr, 2015-10-07:
> Well, it's definitely not part of Verilog 2005. But it is part of SystemVerilog 2012 and seems to be well-defined.
Weird I did not think that SystemVerilog for Vivado was enabled by default.

Probably I could use `localparam` as well. I am fairly inexperienced with Verilog and as I've picked it up quite quickly, I probably have developed one or more bad habits along the way. Basically what I am doing is following: I have a unit with a UART serial interface with N-bit instructions. As I continue to extend the code, I add more and more instructions which in turn means number of bits to describe an instruction increases - to avoid spreading instruction codes and instruction format all over the code, I have nicely gathered this in one file. Feel free to come with suggestions on how you would go about this.

troelsfr, 2015-10-07:
Just a last comment on this one:

Vivado seems to be using SystemVerilog 1800-2009 and for Verilator the code compiles with SV 1800-2005, but not with 1364-2005. Verilator does at no point complain about the global scope parameters, but only other things.