YosysHQ/yosys #89: Vivado+Verilator and Yosys syntax inconsistency

troelsfr, 2015-10-07:
Hey,

Here is another inconsistency: Verilator and Vivado allows you to optionally end an begin/end section with a semicolon:

```
module top(A, clk, rst);
   input clk, rst;   
   output A;   

   always @(posedge clk, posedge rst) begin
      A <= '0';      
   end;  // << like this

endmodule 
```

This causes a syntax error using Yosys and works if the ; is removed. 

Probably this is left to the language developer to chose whether ; is a syntax error, so please feel free to dismiss this "bug" if it is tedious to change.

troelsfr, 2015-10-07:
It may also be a difference in versions of SystemVerilog as you point out in the other thread.

cliffordwolf, 2015-10-08:
I would assume that the two tools that you mentioned parse this as an empty module item, *not* an optional semicolon at the end of the statement. However, I don't think an empty module item is part of either Verilog or SystemVerilog.

Also: `'0'` is *not* a valid Verilog expression. The strings `0` and `"0"` both would be..

I thought I'd add support for empty module items and just print a warning for them, but as it turns out, at least the naive way of doing that would introduce reduce/reduce conflicts into the Yosys Verilog parser. So I'm not adding support for that..

closed 2015-10-08 by cliffordwolf

troelsfr, 2015-10-08:
First of, sorry about '0', it should just have been 0, or 1'b0 - I wrote this minimal example from a more complicated one without testing it before submitting the ticket.

I am not sure what you mean by an "empty module item". Whether or not I have a semicolon at the "end" keyword, the code works and it behaves the same in both case, hence, the phrasing optional.  Just to be sure that you understood me correctly here are the two versions:

```
module top(A, clk, rst);
   input clk, rst;   
   output A;   

   always @(posedge clk, posedge rst) begin
      A <= 0;      
   end;  
endmodule 
```
vs.
```
module top(A, clk, rst);
   input clk, rst;   
   output A;   

   always @(posedge clk, posedge rst) begin
      A <= 0;      
   end
endmodule 
```

This is not a reopening of the ticket, I just wanted to make sure that I was clear enough in the first ticket.