YosysHQ/yosys #91: $finish task support

gaomy3832, 2015-10-12:
The README says
```
- The system tasks $finish and $display are supported in initial blocks
  in and unconditional context (only if/case statements on parameters
  and constant values). The intended use for this is synthesis-time DRC.
```

However, [this simple example](https://github.com/gaomy3832/yosys_tests/blob/master/testbench_finish.v) with `read_verilog` command will fail and generate error as
```
ERROR: System task `$finish' executed at testbench_finish.v:8.
```

It doesn't provide me any useful information about why it fails. So what is the expected result for $finish? Is there any fundamental difficulty to handle $finish?

cliffordwolf, 2015-10-13:
Printing this error is exactly the expected result for `$finish`. The following example demonstrates the how it can be used:

    module test(...);
      parameter GPIO_COUNT = 16;
      ...    
      initial begin
        if (GPIO_COUNT < 0 || GPIO_COUNT > 16) begin
          $display("Parameter Error: GPIO_COUNT must be in range 0..16");
          $finish;
        end
      end
      ...    
    endmodule

closed 2015-10-13 by cliffordwolf

gaomy3832, 2015-10-13:
In your example, $finish is used as an abort command when parameters etc. are illegal. Then it makes sense to terminate the program during compiling.

In my case, $finish is used for terminating the testbench simulation. As you can see in my provided example, we only simulate 100 time units (`#100 $finish;`, the other logic was omitted in the initial block). In this case $finish should only terminate during evaluation, not compilation. Does that mean I should use something else like $stop instead of $finish here?

cliffordwolf, 2015-10-13:
I don't really understand why you are trying to *synthesize* your testbench. In synthesis, all `#`-delays are ignored, so your example effectively contains `initial $finish;`.

gaomy3832, 2015-10-13:
Hmm, I see your point. I imported the whole source tree by `read_verilog`, which happened to include the testbench files. Afterwards I call `hierarchy` to specify the top design.

I just took another look at the document, and used `-defer` option in `read_verilog` to avoid this problem. Thanks!