YosysHQ/yosys #92: clean and opt passes remove blackbox cells not connected indirectly to a top port

Laksen, 2015-10-13:
I would assume the below should be pretty much kept as is no matter what passes I run

```
module top(inout x);

  wire y;

  btestmod1 m1(.a(x), .b(y));
  btestmod1 m2(.a(x), .b(y));

endmodule

(* blackbox *)
module btestmod1(inout a, inout b);

endmodule
```

Laksen, 2015-10-13:
And of course I copy paste the test that actually worked. This is the real issue:

```
module top();

  wire x, y;

  btestmod1 m1(.a(x), .b(y));
  btestmod1 m2(.a(x), .b(y));

endmodule

(* blackbox *)
module btestmod1(inout a, inout b);

endmodule
```

cliffordwolf, 2015-10-13:
In order to keep circuitry that is not driving directly or indirectly any outputs, inouts, (or `assert`/`assume` statements), either mark the instances as "keep":

    module top();
      wire x, y;
      (* keep *) btestmod1 m1(.a(x), .b(y));
      (* keep *) btestmod1 m2(.a(x), .b(y));
    endmodule
    
    (* blackbox *)
    module btestmod1(inout a, inout b);
    endmodule

or the relevant wires:

    module top();
      (* keep *) wire x, y;
      btestmod1 m1(.a(x), .b(y));
      btestmod1 m2(.a(x), .b(y));
    endmodule
    
    (* blackbox *)
    module btestmod1(inout a, inout b);
    endmodule

or the blackbox module itself:

    module top();
      wire x, y;
      btestmod1 m1(.a(x), .b(y));
      btestmod1 m2(.a(x), .b(y));
    endmodule
    
    (* blackbox, keep *)
    module btestmod1(inout a, inout b);
    endmodule

The `hierarchy` pass automatically marks modules as "keep" that contain `assert` and/or `assume` statements.

closed 2015-10-13 by cliffordwolf