YosysHQ/yosys #95: Another fuzzing segfault

jeremysalwen, 2015-10-23:
yosys -S file.v
```
module a(b);inout b=0==c;assign c=^K;assign c=9^k;integer c#0;endmodule
```

```
-- Parsing `/tmp/min.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `/tmp/min.v' to AST representation.
Generating RTLIL representation for module `\a'.
Warning: Identifier `\K' is implicitly declared at /tmp/min.v:1.
Warning: Identifier `\k' is implicitly declared at /tmp/min.v:1.
Successfully finished Verilog frontend.

-- Running command `synth' --

2. Executing SYNTH pass.

2.1. Executing HIERARCHY pass (managing design hierarchy).

2.2. Executing PROC pass (convert processes to netlists).

2.2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

2.2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Removed a total of 0 dead cases.

2.2.3. Executing PROC_INIT pass (extract init attributes).

2.2.4. Executing PROC_ARST pass (detect async resets in processes).

2.2.5. Executing PROC_MUX pass (convert decision trees to multiplexers).

2.2.6. Executing PROC_DLATCH pass (convert process syncs to latches).

2.2.7. Executing PROC_DFF pass (convert process syncs to FFs).

2.2.8. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

2.3. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \a..

2.4. Executing CHECK pass (checking for obvious problems).
checking module a..
Warning: multiple conflicting drivers for a.\c [31]:
    port Y[31] of cell $reduce_xor$/tmp/min.v:1$2 ($reduce_xor)
    port Y[31] of cell $xor$/tmp/min.v:1$3 ($xor)
Warning: multiple conflicting drivers for a.\c [30]:
    port Y[30] of cell $reduce_xor$/tmp/min.v:1$2 ($reduce_xor)
    port Y[30] of cell $xor$/tmp/min.v:1$3 ($xor)
Warning: multiple conflicting drivers for a.\c [29]:
    port Y[29] of cell $reduce_xor$/tmp/min.v:1$2 ($reduce_xor)
    port Y[29] of cell $xor$/tmp/min.v:1$3 ($xor)
Warning: multiple conflicting drivers for a.\c [28]:
    port Y[28] of cell $reduce_xor$/tmp/min.v:1$2 ($reduce_xor)
    port Y[28] of cell $xor$/tmp/min.v:1$3 ($xor)
Warning: multiple conflicting drivers for a.\c [27]:
    port Y[27] of cell $reduce_xor$/tmp/min.v:1$2 ($reduce_xor)
    port Y[27] of cell $xor$/tmp/min.v:1$3 ($xor)
Warning: multiple conflicting drivers for a.\c [26]:
    port Y[26] of cell $reduce_xor$/tmp/min.v:1$2 ($reduce_xor)
    port Y[26] of cell $xor$/tmp/min.v:1$3 ($xor)
Warning: multiple conflicting drivers for a.\c [25]:
    port Y[25] of cell $reduce_xor$/tmp/min.v:1$2 ($reduce_xor)
    port Y[25] of cell $xor$/tmp/min.v:1$3 ($xor)
Warning: multiple conflicting drivers for a.\c [24]:
    port Y[24] of cell $reduce_xor$/tmp/min.v:1$2 ($reduce_xor)
[... 117 lines cut ...]
208	
(gdb) back
#0  (anonymous namespace)::WreduceWorker::run_cell (this=0x7fffffffc630, cell=0x2038db0)
    at passes/opt/wreduce.cc:208
#1  0x000000000174a475 in run (this=<optimized out>) at passes/opt/wreduce.cc:275
#2  (anonymous namespace)::WreducePass::execute (this=<optimized out>, args=..., 
    design=<optimized out>) at passes/opt/wreduce.cc:362
#3  0x00000000004ae8bf in Yosys::Pass::call (design=0x20210a0, args=warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.: 

std::vector of length 32, capacity 32) at kernel/register.cc:227
#4  0x00000000004ace99 in Yosys::Pass::call (design=0x20210a0, command=...) at kernel/register.cc:207
#5  0x0000000001bd0651 in (anonymous namespace)::SynthPass::execute (this=<optimized out>, args=..., 
    design=0x20210a0) at techlibs/common/synth.cc:185
#6  0x00000000004ae8bf in Yosys::Pass::call (design=0x20210a0, args=warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.: 

std::vector of length 32, capacity 32) at kernel/register.cc:227
#7  0x00000000004ace99 in Yosys::Pass::call (design=0x20210a0, command=...) at kernel/register.cc:207
#8  0x0000000000745b0c in Yosys::run_pass (command=..., design=<optimized out>) at kernel/yosys.cc:849
#9  0x00000000004a0276 in main (argc=<optimized out>, argv=<optimized out>) at kernel/driver.cc:363
(gdb) 
```

cliffordwolf, 2015-10-24:
Fixed in commit 2a0f577f839bc43a5045ddf5a2b580bba8cabeab.

That was a nice catch! Keep 'em coming..

closed 2015-10-24 by cliffordwolf

jeremysalwen, 2015-10-24:
This fails with

```
ERROR: Assert `!conn.first.has_const()' failed in kernel/rtlil.cc:1452.
```
would it be possible to also output a message "Cannot assign to a constant" or something?

cliffordwolf, 2015-10-24:
Oops. Fixed it in `wreduce` but broke it in `opt_const`. Fixed fix in commit a1c3df7.