YosysHQ/yosys #98: Parameters in module's function give Verilog read error.

Kmanfi, 2015-11-10:
>Yosys 0.5+     385 (git sha1 ddf3e2d, clang 7.0.0 -fPIC -Os)
>yosys> read_verilog bug_r.v 
>1. Executing Verilog-2005 frontend.
>Parsing Verilog input from `bug_r.v' to AST representation.
>ERROR: Parser error in line bug_r.v:2: syntax error

Example code.
```
module bug_mika(a,z);
   input [7:0] a;
   output [7:0] z;
   parameter pos = 1;

   assign z = ff(a);

   function [7:0] ff;
      parameter pos2 = pos + 1;
      input [7:0] arg1;
      begin
        ff = arg1[pos2:0];
      end
   endfunction
endmodule  
```

cliffordwolf, 2015-11-11:
Thanks for bringing this to my attention! This is now fixed by commit 34f2b84fb60445210ff9ffdf33b90ef6f50b91ed.

closed 2015-11-11 by cliffordwolf

Kmanfi, 2015-11-18:
Cheers. :)