YosysHQ/yosys #102: Incompatible re-declaration of wire

Kmanfi, 2015-11-18:
```
Yosys 0.5+     393 (git sha1 fd3e10c, clang 7.0.0 -fPIC -Os)

yosys> read_verilog bug_1.v 
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `bug_1.v' to AST representation.
Generating RTLIL representation for module `\test'.
ERROR: Incompatible re-declaration of wire $func$\muxer$bug_1.v:19$2$\size at bug_1.v:2.
```

```
module test (
  a, b
);
 
  input a;
  output [1535:0] b;

  reg [1535:0] G;
  reg F;
  reg H;
  reg I;
  reg J;

  reg [5:0] K;
  reg [9:0] L;
  reg [9:0] M;

  assign b = muxer(G, {L , H , F , M, J , I}, K, 24, 0);

  function [1535:0] muxer;
    input [1535:0] vector;
    input [23:0] slice;
    input [5:0] index;
    input size;
    integer size;
    input offset;
    integer offset;
    integer i;
    reg [1535:0] muxed_value;
  begin
    muxed_value = vector;
    for (i = 0; i < 24; i = i+1)
      muxed_value[index * size + offset + i] = slice[i];
    muxer = muxed_value;
  end
  endfunction

endmodule
```

Kmanfi, 2015-11-21:
Above code seems to work with iverilog.

cliffordwolf, 2015-11-23:
Fixed the error in commit c86fbae. However, there also seems to be a performance bug (your example needs ~4 minutes to process on my machine). I'll be looking into that later this week..

Kmanfi, 2015-11-24:
This doesn't work but...
```
module data_test(a, z);
   input [7:0] a;
   output [7:0] z;  
   reg  [7:0] z;

   always@(a)
     begin
       do_data(a,z);
     end

   task do_data;
      input [7:0] data_in;
      output [7:0] data_out;
      reg [7:0] data;

      begin
	data_out = 0;
      end
   endtask

  endmodule
```
This works.
```
module data_test(a, z);
   input [7:0] a;
   output [7:0] z;  
   reg  [7:0] z;

   always@(a)
     begin
       do_data(a,z);
     end

   task do_data;
      input [7:0] data_in;
      output [7:0] data_out;
      reg [7:0] data;

      begin
	data_out = 0;
       // by adding only this comment, this works.
      end
   endtask
  endmodule
```

cliffordwolf, 2015-11-24:
Hmm.. They both work fine here with `Yosys 0.5+398 (git sha1 8ff229a, clang 3.4-1ubuntu3 -fPIC -Os)`. Valgrind does not complain either. What is the error message you are getting?

Kmanfi, 2015-11-24:
My version was same. I made.
-make clean 
-make -j8
Now it works.

Kmanfi, 2015-11-30:
There is the performance bug still left. It takes about 4min. to read Verilog. proc; opt takes many days.

cliffordwolf, 2015-12-05:
Just a little update from my side:

The problem here is that the RTLIL representation generated by the Verilog front-end for this test case is very bad. In the long run I'll have to improve the Verilog front-end, but I probably won't do this for a while because it is complicated and I don't have the time now..

However, it is interesting to see what the rest of the flow does with the bad RTLIL code generated by the front-end. (It is the equivalent of a single always block with ~100k lines of code.)

There was a performance bug in `proc_mux` that I have fixed now. Now it would complete in a few hours, but unfortunately it runs out of memory about a quarter into the example (it would take approx. 40 GB of RAM to complete `proc_mux` on that RTLIL code).

I've now created a variation of your code that uses a vector length of 500 instead of 1536. That "only" needs 6 GB of RAM to process. With the output of that I can see that there also is a performance bug in `proc_dlatch`. I am planning on fixing that one next.

Ultimately I need to fix the Verilog front-end of course, but I think it will be 2016 until I get to that. Until then you could write code that closer resembles what the Verilog front-end should do with that design. Something like the following:

    begin
      muxed_value = vector;
      for (i = 0; i < 24; i = i+1) begin
        if (slice[i])
          muxed_value = muxed_value |  (1 << (index * size + offset + i));
        else
          muxed_value = muxed_value & ~(1 << (index * size + offset + i));
      end
      muxer = muxed_value;
    end

However, there seems to be yet another performance bug in the `share` pass that is exposed by this variant of the code.. (What is it with this test case!? ;)

So long story short: I am working on this. But it might take a while because this test case exposes multiple problems at once and I addressing them in the order that fixes the most bugs in yosys, not necessarily the order that makes this one example work asap.

I'll post updates here with my progress. But expect it to take a while..

Kmanfi, 2015-12-05:
Thank you for the update.

cliffordwolf, 2016-04-21:
I have now fixed the performance bug in the `share` pass in commit 1565d1af69f552b790aa43fd6be194ee59ab76f3. So now the version of the test case with

    begin
      muxed_value = vector;
      for (i = 0; i < 24; i = i+1) begin
        if (slice[i])
          muxed_value = muxed_value |  (1 << (index * size + offset + i));
        else
          muxed_value = muxed_value & ~(1 << (index * size + offset + i));
      end
      muxer = muxed_value;
    end

instead of

    begin
      muxed_value = vector;
      for (i = 0; i < 24; i = i+1)
        muxed_value[index * size + offset + i] = slice[i];
      muxer = muxed_value;
    end

passes `proc; opt; share`. however, trying to run `synth` eventually gets stuck on the same `proc_dlatch` performance bug as the other test case, this time triggered by RTLIL code generated internally for the wide shift cells during lowering to a gate-level representation.

Thanks for your patience. I'll keep you posted on my progress.

cliffordwolf, 2016-04-24:
I've now fixed the `proc_dlatch` performance bug in commit b1d6f05fa2156017f50383d01d49342c8ec5e209. My alternative version of the code with large bit shifts instead of indexed addressing now runs though `synth` in about one minute.

cliffordwolf, 2016-04-25:
With commit 93e107e455b506731d9114e0dc2644f78797cf0f I can now process your original verilog code in about 15 minutes and with less than 2 GB of RAM. Not perfect, but at least possible to process.

With that I'm now finally closing this issue. There is still work to be done in the Verilog front-end to generate better RTLIL code in the first place. There are plans for some major refactoring of the Verilog front-end in the long term that might take care of that anyways and I don't want to make changes to it before that unless absolutely necessary. So I guess this is as good as it gets for now..

closed 2016-04-25 by cliffordwolf

Kmanfi, 2016-04-25:
Thank you very much.