YosysHQ/yosys #104: Memory exhausted

Kmanfi, 2015-11-27:
How can I help with this?

```
 Yosys 0.5+     402 (git sha1 a7ffb85, clang 7.0.0 -fPIC -Os)
yosys> read_verilog big_test.v
...
Successfully finished Verilog frontend.
yosys> write_ilang tmp.ilang 

3. Executing ILANG backend.
Output filename: tmp.ilang

yosys> design -reset

yosys> read_ilang tmp.ilang 

4. Executing ILANG frontend.
Input filename: tmp.ilang
ERROR: Parser error in line 247340: memory exhausted
```

cliffordwolf, 2015-11-27:
Please provide the verilog input and yosys script, or `tmp.ilang`. How large is `tmp.ilang`? I need to be able to reproduce the problem in order to be able to do anything..

cliffordwolf, 2015-11-27:
Fyi: I have now [found and fixed a performance issue in the ilang front-end](https://github.com/cliffordwolf/yosys/commit/32f5ee117cd419e35db9f7ea15b958e1d545ecea), but I don't think it is related to the problem you are seeing here.

Kmanfi, 2015-11-27:
I tested the big_test.v with Yosys 403 version and no difference. I am currently isolating this issue to smaller test case. Current size of tmp.ilang is 16115174.

cliffordwolf, 2015-11-28:
16 MB is not so bad. Can you upload that ilang file somewhere (Google Drive, Dropbox, ..) and post the link here? Or zip it and mail it to clifford@clifford.at.

Kmanfi, 2015-11-29:
I mailed this to clifford@clifford.at.

cliffordwolf, 2015-11-29:
Thanks! Got it. Fixed in commit 4a697ac.

closed 2015-11-29 by cliffordwolf