YosysHQ/yosys #106: Undefined subcircuits in verilog-> translation

jlanik, 2015-12-02:
Hi,

I want to generate a .blif for a verilog design.
I used yosys -o <blif-file> -S <verilog-files>

The resulting .blif contains some undefined black boxes, 
example:
.subckt $_DFF_PN0_ C=clk D=$0\rp[1:0][0] Q=rp[0] R=rst

It seems to me, that those are some type of flip-flops, however the definition is missing.
It seems to me, that this problem comes from the translation, there are no such black-box FFs in the original design.

Thank you

cliffordwolf, 2015-12-02:
Your design contains FFs with asynchronous resets. (Namely FFs with positive edge clock and negative edge reset to zero.) These do not translate to the set of supported FF types in BLIF (see section 4 of the [BLIF spec](https://www.ece.cmu.edu/~ee760/760docs/blif.pdf)). So yosys outputs a `$_DFF_PN0` cell, the internal cell type in Yosys used to represent this type of FF.

[Yosys Appnote 10](http://www.clifford.at/yosys/files/yosys_appnote_010_verilog_to_blif.pdf) discusses this issue and a possible work-around briefly in section IV (changing the async reset to a sync reset in the synthesis script).

(A simpler solution would of course be to change your HDL code to use sync resets. But depending on your flow this might not be a feasible option.)

closed 2015-12-02 by cliffordwolf

jlanik, 2015-12-02:
I see, thank you.