YosysHQ/yosys #107: Help needed setting up a PLL

carlosgs, 2015-12-05:
Hi Clifford, first of all thanks for your great work with the open FPGA toolchain (and with OpenScad!) :-)

I'm trying to set up a PLL with the ICEstick board. For the sake of simplicity the first attempt is to make an LED blink at Fclk/8 (DIVR=7, DIVF=0, DIVQ=0).

[image]  
(from the [Lattice docs](http://www.latticesemi.com/~/media/LatticeSemi/Documents/ApplicationNotes/IK/iCE40sysCLOCKPLLDesignandUsageGuide.pdf?document_id=47778) page 7)

Here is the code I'm working with:
```verilog
module program(
    input   clk, // set_io clk 21
    output  LED // set_io LED 99
);

    wire clk;
    wire LED;

    wire clk_new;
    wire BYPASS;
    wire RESETB;

    SB_PLL40_CORE #(
        .FEEDBACK_PATH("PHASE_AND_DELAY"),
        .DELAY_ADJUSTMENT_MODE_FEEDBACK("FIXED"),
        .DELAY_ADJUSTMENT_MODE_RELATIVE("FIXED"),
        .PLLOUT_SELECT("SHIFTREG_0deg"),
        .SHIFTREG_DIV_MODE(1'b0),
        .FDA_FEEDBACK(4'b0000),
        .FDA_RELATIVE(4'b0000),
        .DIVR(4'b1111),
        .DIVF(7'b0000000),
        .DIVQ(3'b000),
        .FILTER_RANGE(3'b001),
    ) uut (
        .REFERENCECLK   (clk),
        .PLLOUTCORE     (clk_new),
        .BYPASS         (BYPASS),
        .RESETB         (RESETB)
        //.LOCK (LOCK )
    );

    assign LED = clk_new;
    assign BYPASS = 0;
    assign RESETB = 0;

endmodule
```

Unfortunately the LED remains fully off. I've tried setting BYPASS=1 and then the LED is visibly on.
It probably is something dumb... but as a newbie I cannot spot it, so your help would be very much appreciated.

Also, I wanted to check with you whether it is normal that the clock input is not assigned to a Global Buffer by default:  
[image]  
What is the proper way to force CLK to be a GB?

Thanks plenty,
Carlos

carlosgs, 2015-12-05:
Hmm I just realized that my version of the FPGA toolchain is a bit outdated (~1.5 months old), I'll update it tomorrow and check for any differences.

cliffordwolf, 2015-12-05:
There is a multitude of problems with the code you posted:

1. The `RESETB` signal is inverted (thus the `B`). So it must be set to `1` in order for the PLL to run.

2. The valid range for `DIVQ` is 1..6. This is correctly documented in the [ICE Technology Library Document](http://www.latticesemi.com/~/media/LatticeSemi/Documents/TechnicalBriefs/SBTICETechnologyLibrary201504.pdf), but unfortunately incorrectly documented as 0..7 in the iCE40 sysCLOCK PLL Design and Usage Guide.

3. After setting `RESETB=1` and `DIVQ=1` the PLL will produce a signal, but it [will not be a square wave](https://www.instagram.com/p/-54c3SK4Hz/). The valid PLL output range is 16 - 275 MHz. You are trying to create a 12 / 16 = 750 kHz signal. That is not possible. Also note that the valid PLL VCO range is 533 - 1066 MHz (!), so you must choose your parameters so that the VCO frequency is in that range. You are trying to run the VCO at 750 kHz as well, approx. 1000 times slower than the lower bound of its operating range! Finally you also trying to run the phase detector at 750 kHz. But its operating range is 10 - 133 MHz. (see f_OUT, f_VCO and f_PFD on page 3-16 (sysCLOCK PLL Timing) in the [iCE40 LP/HX Family Data Sheet](http://www.latticesemi.com/~/media/LatticeSemi/Documents/DataSheets/iCE/iCE40LPHXFamilyDataSheet.pdf?document_id=49312).

> Also, I wanted to check with you whether it is normal that the clock input is not assigned to a Global Buffer by default:

You are not using it as a clock anywhere in the design. Drive the clock input of FFs with it and arachne-pnr will route the signal through a clock buffer. Alternatively you can manually instantiate a clock buffer (SB_GB or SB_GB_IO) for a signal. When using PLLs you need the PLLOUTGLOBAL output if you want to use the direct path from the PLL to the global clock distribution network.

closed 2015-12-05 by cliffordwolf

carlosgs, 2015-12-05:
Thanks for your quick and detailed reply, it would have taken me ages to figure out all of those matters... Also it is very interesting to see the waveforms since I don't have an oscilloscope at home.

Got the LED to blink at ~1.072Hz using 12MHz*3/(2^25) (with a 25bit prescaler):  
<https://github.com/carlosgs/iceDAQ/blob/master/pll_example/pll_example.v>

I still need to understand how the VCO parameters (DIVQ & FILTER_RANGE) are properly determined, but at least the PLL is now working, thanks so much!!!
Again, great work Clifford, it is amazing to be able to do all this only with open tools :-)

AntonKrug, 2019-11-17:
Sorry for reviving an old thread, but just in case google search will lead people here who where searching how to configure PLL. The *icepll* is in the Clifford's IceStorm project for about ~3 years. It can generate the config values and even output a verilog module to be used with yosys.