YosysHQ/yosys #109: possible regression: 5e1dc78 uses 98 more LCs than 9f5b6e4

RGD2, 2015-12-08:
The design is the j1a from jamesbowman/swapforth@9d45735 , and the problem is that 98 more LCs is 17 too many to fit into the ice40hx1k, which breaks that design. ( jamesbowman/swapforth#18 )

It might be that 9f5b6e4 is producing a slightly broken BLIF and 5e1dc78 is producing a more correct one, in which case perhaps the swapforth j1a should be refactored to fit with the new yosys.

cliffordwolf, 2015-12-08:
5e1dc78 is not a yosys commit. I've made a bisect starting with a month back. The bad commit is 1ea6db3. I'll look into it. However, there is a chance that this commit just changed the behavior of `proc_mux` in a "neutral" way, causing the overall output to improve in some cases to worsen and in others randomly..

cliffordwolf, 2015-12-08:
I've now finished checking: Commit 1ea6db3 only changes `proc_mux` and the output of proc mux for the J1a is structurally equivalent before and after that commit. (I.e. the only thing that changed are net and cell names.)

In commit 494e5f2 I have now added the option -abc2 to "synth_ice40" that runs logic optimization twice. This improves the synthesis results a bit. With this I can implement the j1a with 1225 / 1280 LC with yosys git head.

closed 2015-12-08 by cliffordwolf

RGD2, 2015-12-08:
Thanks Clifford! Much obliged :)