YosysHQ/yosys #112: memory instantiation takes a long time and consume a lot of memory

lynxis, 2015-12-22:
Hi,

I'm trying to use memory with dual ports, but it ran into the OOM after taking >4gb before it got killed by OOM.

```
time yosys -Q -p "synth_ice40 -blif buffer.blif" buffer.v  > buffer.log
Killed

real	10m23.361s
user	10m17.600s
sys	0m4.860s
```

[buffer.log.bz2](https://own.fe80.eu/index.php/s/upJEfO1jtKSmJcH)

Yosys 0.5+411 (git sha1 9df59f0, clang 3.7.0 -march=x86-64 -mtune=generic -O2 -fstack-protector-strong -fPIC -Os)

buffer.v:
```verilog
module buffer #(parameter AW = 16, parameter DW = 8)
        (
                input write_clock,
                input [DW-1:0] write_data,
                input [AW-1:0] write_addr,

                output [DW-1:0] read_data,
                input [AW-1:0] read_addr);

        localparam NPOS = 2 ** AW;

        reg [DW-1: 0] ram [0: NPOS-1];

        always @(posedge write_clock)
                ram[write_addr] <= write_data;

        always @(read_addr)
                read_data <= ram[read_addr];
endmodule
```

cliffordwolf, 2015-12-22:
iCE40 block RAMs only support synchonous reads. I.e. something like this:

```verilog
module buffer #(parameter AW = 16, parameter DW = 8)
        (
                input write_clock,
                input read_clock,
                input [DW-1:0] write_data,
                input [AW-1:0] write_addr,

                output [DW-1:0] read_data,
                input [AW-1:0] read_addr);

        localparam NPOS = 2 ** AW;

        reg [DW-1: 0] ram [0: NPOS-1];

        always @(posedge write_clock)
                ram[write_addr] <= write_data;

        always @(posedge read_clock)
                read_data <= ram[read_addr];
endmodule
```

Memories with asynchrous reads would need to be built from discrete logic and FFs. Yosys is trying to do that for you, but we are talking about `(2**16)*8` = half a million FFs here and at least a few million LUTs for the address logic, a couple of thousand times larger than the largest iCE40 FPGA.

closed 2015-12-22 by cliffordwolf

lynxis, 2015-12-23:
thanks for your fast answer.