YosysHQ/yosys #117: clang crash on ubuntu 14.04

jeremyherbert, 2016-02-03:
Hi,

I'm trying to install yosys on a brand new vagrant box, but clang seems to crash at the same place every time. Is this a known issue? I'll probably still flag it with the debian guys anyway.

Output is as follows:
```
$ make
[ 57%] Building passes/opt/opt_const.o
clang: error: unable to execute command: Killed
clang: error: clang frontend command failed due to signal (use -v to see invocation)
Ubuntu clang version 3.4-1ubuntu3 (tags/RELEASE_34/final) (based on LLVM 3.4)
Target: x86_64-pc-linux-gnu
Thread model: posix
clang: note: diagnostic msg: PLEASE submit a bug report to http://bugs.debian.org/ and include the crash backtrace, preprocessed source, and associated run script.
clang: note: diagnostic msg:
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang: note: diagnostic msg: /tmp/opt_const-02e5ad.cpp
clang: note: diagnostic msg: /tmp/opt_const-02e5ad.sh

clang: note: diagnostic msg:

********************
make: *** [passes/opt/opt_const.o] Error 254
```

Diagnostic files here: [opt_const-02e5ad.zip](https://github.com/cliffordwolf/yosys/files/115396/opt_const-02e5ad.zip)

Vagrantfile:
```
Vagrant.configure(2) do |config|
  config.vm.box = "ubuntu/trusty64"

   config.vm.provision "shell", inline: <<-SHELL
     sudo apt-get update
     sudo apt-get install -y build-essential clang bison flex libreadline-dev gawk tcl-dev libffi-dev git mercurial graphviz xdot pkg-config python python3 libftdi-dev
   SHELL
end
```

cliffordwolf, 2016-02-03:
Is it possible that you run out of memory in your VM? On my machine it takes almost 700 MB of resident memory to build this file:

```
$ rm passes/opt/opt_const.o
$ /usr/bin/time -v make passes/opt/opt_const.o
...
Maximum resident set size (kbytes): 691616
```

jeremyherbert, 2016-02-04:
Yep, memory was the problem. My bad. Thanks for your help.

Here is the vagrant setup in case anyone is interested: https://github.com/jeremyherbert/icestorm-vagrant

closed 2016-02-04 by jeremyherbert