YosysHQ/yosys #118: DDR output cell simulation

alibabashack, 2016-02-06:
I might have found a bug in the ice40 techlib for SB_IO (yosys/techlibs/ice40/cells_sim.v):
According to the Lattice spec TN1253 ([1] page 11) the two flip flops in the IO cell used for DDR output mode are sensitive to the rising and falling clock edge respectively. The output of each flip flop is fed into a multiplexer controlled by the clock, such that each flip flop drives the pin for a half clock cycle. From the document it is not clear which flip flop drives the output in which phase. The current model switches the multiplexer to the flip flop which just updated its output. In my Icarus simulation this results in glitches.  I think this multiplexer's control input must be inverted in the model.

I have not tested on an FPGA, but in another Lattice document for a different FPGA series ([2] page 11, figure 10/11) the inverted sense of the multiplexer is documented. This also makes perfect sense for me.

A fix which works for me is to swap the dout assignment to
`dout = (OUTPUT_CLK ^ NEG_TRIGGER) || PIN_TYPE[2] ? dout_q_1 : dout_q_0;`

[1] http://www.latticesemi.com/view_document?document_id=47960
[2] http://www.latticesemi.com/~/media/LatticeSemi/Documents/ApplicationNotes/MO/ORCASeries4FastDDRInterface.PDF?la=en

cliffordwolf, 2016-02-07:
Afaict the current HDL code matches the schematic on page 87 of the [iCE40 technology library documentation](http://www.latticesemi.com/~/media/LatticeSemi/Documents/TechnicalBriefs/SBTICETechnologyLibrary201504.pdf). I might be mistaken so please double check..

However, I can see how this glitches in simulation: the `always @*` that updates `dout` will be updated blocking at a clock edge to drive the old value of `dout_q_*`. Only after that the nonblocking update of `dout_q_*` happens, the always block for `dout` runs again and updates `dout` to the new value of `dout_q_*`.

I think I have now [fixed this in git head](https://github.com/cliffordwolf/yosys/commit/0ccfb88728c5bdf167b1d672034ca5cf360dedb1). Please test and let me know if this solution works for you.

alibabashack, 2016-02-08:
> I might be mistaken so please double check..

Looks good indeed.

I read your explanation for the glitches several times and I'm still puzzled ;-) However, your fix works for me; all signals look as expected.

Thank you!

closed 2016-02-08 by alibabashack

cliffordwolf, 2016-02-08:
> I read your explanation for the glitches several times and I'm still puzzled.

If you are still interested: Maybe this SNUG-2000 Cliff Cummings paper helps explain the effect of blocking and nonblocking assignments and the semantics of the Verilog event queue model.

http://www.sunburst-design.com/papers/CummingsSNUG2000SJ_NBA.pdf

(But maybe you don't like my fix that much anymore after you figured out what the problem was and how I worked around it. It's a bit ugly.. ;)

RGD2, 2016-02-09:
Thanks Clifford, nice reference on blocking/nonblocking assignment there! I found it helpful.

alibabashack, 2016-02-09:
That was a good read! I noticed the ugliness of the fix, but now I understand what's going on there. Thanks!