YosysHQ/yosys #121: more documentation on synthesized primitives, please

klammerj, 2016-03-01:
I'd like to know what the $macc $alu and $lcu nodes do. 
Also, $_DLATCH_ variants.

cliffordwolf, 2016-03-01:
Type e.g. `help $macc` or `help $_DLATCH_P_` on the yosys command prompt (or run e.g. `yosys -h '$macc'` from a shell) to display the help message for a cell type. Not all cell types have a help message yet (e.g. most of the lowercase named "RTL" type cells don't; this is work in progress), but all cell types have a Verilog simulation model that can be displayed e.g. with `help $macc+` (as indicated by the `help $macc` output).

I'll leave this ticket open as a reminder to continue writing cell type help messages..

povik, 2023-09-29:
As of today there's only a few coarse cells that have help messages, most don't. On the other hand the internal gate cells all seem to have a help text. Help texts for cells are parsed out from `techlibs/common/simcells.v` and `.../simlib.v`.

NachtSpyder04, 2024-08-14:
Hello, 
I would like to work on this issue.

Can anyone guide me which cells should I target first?

povik, 2024-08-14:
I would suggest to start with the simple ones (equality operators `$eq`, `$neq`, arithmetic comparison `$lt` and friends, `$logic_*` operators).

KrystalDelusion, 2025-02-14:
@akashlevy except for [$fa](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cell/word_arith.html#arith.$fa), [$lut](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cell/word_logic.html#logic.$lut), [$fsm](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cell/word_fsm.html#fsm.$fsm), [$sop](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cell/word_logic.html#logic.$sop), [specify rules](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cell/word_spec.html), [formal cells](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cell/word_formal.html), [debug cells](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cell/word_debug.html), and [$slice](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cell/word_wire.html#wire.$slice).  Also open todos $_ALDFFE?_[NP]{2,3}_, $alu, $lcu, and $concat.  And that's just the ones with no documentation, I suspect there are other cells/groups that need a refresh too.

akashlevy, 2025-02-14:
Got it!

widlarizer, 2025-02-14:
`$sop` and `$lut` are documented, right above where the link from @KrystalDelusion goes, unless there's some missing piece I'm not aware of

KrystalDelusion, 2025-02-14:
They're read the docs only, nothing shows up if you do `help $sop`. Same for memories and registers.