YosysHQ/yosys #122: Placing problems

Mecrisp, 2016-03-06:
With update from Yosys shapshot from 17. November 2015 to Yosys 0.6 package Mecrisp-Ice 0.3 fails to build with "fatal error: failed to place: placed 1103 LCs of 1257 / 1280". The old version of Yosys synthesized the same design into 1241 LCs and placed properly. Swapforth from James Bowman also now fails to build with the same error.

After I tidied up my design a bit to reduce footprint, I found that the -abc2 option to the synth_ice40 script can actually increase necessary LC count. Is there another "optimize for area" option for Ice40 HX1K in Yosys ? I would love to see a common synthesis script "synth_ice40_area" included in Yosys...

I faced the "failed to place" message a few times before with snapshot from 17. November, which could be solved in the past by logical identical reordering of Verilog constructs. How can I get feedback on what constructs fail to place in order to change this and that in the design to help the place process ?

Matthias

cliffordwolf, 2016-03-06:
This design uses too many of the LUTs in the chip and therefore sometimes it just does not fit. This is not due to any actual behavioral changes but things like different names for cells and nets in different versions of the tools that cause slight fluctuations in things like the order in which cells are processed.

The chip has space for 1280 LCs, but those LCs are grouped in blocks of 8 LCs and signals like SET/RESET/ENABLE/CLOCK are shared between those 8 LCs. Because of this your design can have fewer than 1280 LCs and still not fit.

If there is something that qualifies as bug, it is in arachne-pnr as this is the tool that does packing and placement, not Yosys. So I'm going to close this ticket here. You might want to open a ticket for arachne-pnr if you think there is an actual bug here (or at least a feature request for trying harder to pack and place the design so it fits the chip). If you do file an issue with arachne-pnr, make sure to include files that allow the author to quickly reproduce the problem. For example by including the BLIF and PCF files and the exact arachne-pnr command line.

You can try varying the `-s` parameter for arachne-pnr (the seed to the random number generator) and see if you find a value for which the tool succeeds placing the design.

> How can I get feedback on what constructs fail to place

You are trying to squeeze to much stuff into a too small chip. If you want to seat 5 people on a table for 4 people, it really does not make any sense to ask which one is the one that does not fit. It just too many people and not enough seats.. Of course you can try assigning seats to people and then you would end up with one person without seat, but who that person is is random and does give you any insights into the problem. I hope this analogy helps understanding the problem..

closed 2016-03-06 by cliffordwolf

jamesbowman, 2016-03-06:
Thanks @cliffordwolf, this is clear. We will follow up with arachne-pnr.

I would like to get a better understanding of what structures map efficiently to LCs; are there any ways of inspecting the placement and matching it back to the original code?

cliffordwolf, 2016-03-06:
> are there any ways of inspecting the placement and matching it back to the original code?

you can ask arachne-pnr to produce a post-pack and post-place-and-route netlists (e.g. using `--post-place-blif`, see `--help` for more options). The "src" attributes should be preserved when doing so (so you should be able to relate the cells to the original HDL code) and "loc" attributes should be added by arachne-pnr to indicate the final placement.

Mecrisp, 2016-03-06:
Dear Clifford, thank you for the reply !

We will try whatever possible, and I asked you first as a change in the Yosys revision changed placeability.

Additionally, maybe a different, area-optimized technology mapping step in the synthesis script could help, I found that ABC has two algorithms for that:

mfs – An area-oriented resynthesis engine for network mapped into K-LUTs described in the paper.
lutpack – An area-oriented resynthesis engine for network mapped into K-LUTs described in the paper. 

Matthias

cliffordwolf, 2016-03-07:
@Mecrisp good idea! I've now added a calls to "mfs" and "lutpack" to "abc -lut" in commit c4aaed099f948f821. In my tests this improves synthesis area results a little bit.

Mecrisp, 2016-03-07:
Hi Clifford,

thank you - I tried it and the small gain really makes a difference for me. Now I can place my design.

A very happy Matthias

jamesbowman, 2016-03-08:
Wow, me too! J1a is now fitting happily:
```LCs          1162 / 1280```

larsbrinkhoff, 2017-06-02:
Sorry to butt in, but I'm toying with a Forth CPU and I want it to fit on the HX1K.  (With only 11 instructions, I'd be rather annoyed if it didn't.)

@jamesbowman, is the smallest version of the J1a in your j1 repository, or swapforth?

Mecrisp, 2017-06-02:
Hi Lars,

the smallest is in James "swapforth" folder, the J1a nicely fits into the HX1K.
https://github.com/jamesbowman/swapforth/tree/master/j1a

My fork of Swapforth, Mecrisp-Ice, also fits in HX1K, but it is larger, containing more features like a timer interrupt and msp430 style IO ports - unfortunately building only with an old version of Yosys, as it is a very tight fit.

Ready-made bitstreams are included in both projects.

Matthias