YosysHQ/yosys #123: How can I replace GND and VCC nodes

klammerj, 2016-03-08:
I'd like to dump an edif file with certain elements replaced. 
mostly works. 
replacing VCC and GND with ONE and ZERO is causing me problems, tho'.
having this in a cells_map.v
does not replace them
```
module  GND (output G); ZERO _TECHMAP_REPLACE_ (.Q(G)); endmodule
module  VCC (output P); ONE _TECHMAP_REPLACE_ (.Q(P)); endmodule
```

\$VCC and \$GND doesn't work either. 

help -cells does not seem to show any constants.

more of that mess there: <https://github.com/klammerj/y40k>

klammerj, 2016-03-08:
I've used ```hilomap -hicell ONE Q -locell ZERO Q``` now, and that works. 
Unfortunately there are still instances of GND and VCC in the resulting edif file(one each). They are unconnected. Can I get rid of those? How?

cliffordwolf, 2016-03-08:
I've now added a `-nogndvcc` option to `write_edif` in commit d117893007baa506b702967ac6ee9f70af3c236a that should do the job.

closed 2016-03-08 by cliffordwolf