YosysHQ/yosys #125: Yosys & iverilog differences in defines

chriscamacho, 2016-03-11:
I have a macro
`define INCPC()     PC<=PC+1; romA<=PC+1; IR<=romD

used like this later in an always clause
`INCPC();

This synthesises and works as expected on real hardware (xilinx 8k board)

attempting to get some simulation I can have confidence with I run:
 iverilog -o test_post -D POST_SYNTHESIS atto_tb.v atto.v `yosys-config --datdir/ice40/cells_sim.v`

I get the following
atto.v:7: error: malformed `define directive.
atto.v:123: warning: macro INCPC undefined (and assumed null) at this point.

as well as a number of syntax errors etc I also see a bunch of errors in yoysys's resources

/usr/local/share/yosys/ice40/cells_sim.v:18: error: malformed statement
/usr/local/share/yosys/ice40/cells_sim.v:19: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:19: Syntax in assignment statement l-value.

cliffordwolf, 2016-03-12:
According to IEEE Std. 1364-2005 Syntax 19-2—Syntax for text macro definition (see sec. 19.3.1 \`define), you can either define a macro without arguments and without parentheses, or with arguments and parentheses. I.e. the following two macro definitions are valid Verilog code:

    `define FOO 1234
    `define BAR(arg1, arg2) (arg1 + arg2)

However, macro definitions with empty parentheses are not valid Verilog code. I.e. the following macro definition is invalid:

    `define FOOBAR() 1234

Icarus Verilog is more strict with things like that than Yosys and therefore rejects the macro definition with empty parentheses.

> as well as a number of syntax errors etc I also see a bunch of errors in yoysys's resources

I cannot reproduce this. Which version of Icarus Verilog are you using? (What is the first line in the output of `iverilog -V`?) I have run `ice40/cells_sim.v` through Icarus Verilog 0.9.7 and current git head of Icarus Verilog, and both accepted it.

closed 2016-03-12 by cliffordwolf

chriscamacho, 2016-03-12:
Icarus Verilog version 10.0 (stable) ()  (basically what comes with my distro (void) )
I only see issues (with cells_sim.v) when define is incorrectly used

Yosys should generate a meaningful error such as

> Syntax error define used without parameters should have no brackets

can you re-poen this until Yosys does generate an error message correctly?
I believe better error reporting will improve Yosys significantly

cliffordwolf, 2016-03-12:
Quoting the `read_verilog` help message:

> Note that the Verilog frontend does a pretty good job of processing valid
verilog input, but has not very good error reporting. It generally is
recommended to use a simulator (for example Icarus Verilog) for checking
the syntax of the code, rather than to rely on read_verilog for that.

Also: If you synthesize first, simulate later, then something is seriously wrong with your design work flow.

chriscamacho, 2016-03-12:
struggling to get anything to work at all if I simulate first, they behave too differently...

cliffordwolf, 2016-03-12:
You have not yet posted anything that has different behavior in simulation and synthesis, so I can't comment on that.

But whatever it is, I can ensure you: *If you synthesize first, simulate later, then something is seriously wrong with your design work flow.*