YosysHQ/yosys #126: Duplicate default in switch case.

Kmanfi, 2016-03-14:
When writing verilog, Yosys makes duplicate default clause. Some commercial tools consider this as an error.

```
    casez (in_a)
      2'b00:
        begin
          \$1\result[2:0]  = 3'b101;
        end
      2'b01:
        begin
          \$1\result[2:0]  = 3'b001;
        end
      default:
          \$1\result[2:0]  = 3'b000;
      default:
          \$1\result[2:0]  = 3'b000;
    endcase
```

This will make duplicate default.
```
module test (in_a, out_vt);

input [1:0] in_a;
output out_vt;
reg   [2:0] result;
assign out_vt = result;

always @(*)
  begin
     result    = 3'b000;  
     case (in_a)
       2'b00 : begin
          result = 3'b101;  
       end

       2'b01: begin
          result    = 3'b001;
       end

       default;
 
     endcase
  end
endmodule

```
This works.
```
module test (in_a, out_vt);
input [1:0] in_a;
output out_vt;
reg   [2:0] result;
assign out_vt = result;

always @(*)
  begin
     result    = 3'b000;  
     case (in_a)
       2'b00 : begin
          result = 3'b101;  
       end

       2'b01: begin
          result    = 3'b001;
       end

       default:
        begin
        end

     endcase
  end
endmodule
```

Kmanfi, 2016-03-14:
One variation more.
yosys> read_verilog fifo_3.v 
yosys> hierarchy -top test
yosys> write_verilog -noexpr -norename test.v

Now yosys generates two defaults, with and without "begin ends".
```
    casez (in_a)
      2'b00:
        begin
          \$1\result[2:0]  = 3'b101;
        end
      2'b01:
        begin
          \$1\result[2:0]  = 3'b001;
        end
      default:
        begin
          \$1\result[2:0]  = 3'b011;
        end
      default:
          \$1\result[2:0]  = 3'b000;
    endcase
```

```
module test (in_a, out_vt);

input [1:0] in_a;
output out_vt;

reg   [2:0] result;

assign out_vt = result;

always @(*)
  begin

     result    = 3'b000;  

     case (in_a)
       2'b00 : begin
          result = 3'b101;  
       end

       2'b01: begin
          result    = 3'b001;
       end

       default result = 3'b011;
     endcase
  end
endmodule
```

Kmanfi, 2016-03-14:
Even more advanced version. Commercial tool say "Only one default per case statement allowed".
```
    casez (in_b)
      1'b1:
        begin
          \$1\result[2:0]  = \$2\result[2:0] ;
          casez (in_a)
            2'b00:
              begin
                \$2\result[2:0]  = 3'b101;
              end
            2'b01:
              begin
                \$2\result[2:0]  = 3'b001;
              end
            default:
              begin
                \$2\result[2:0]  = 3'b011;
              end
            default:
                \$2\result[2:0]  = result;
          endcase
        end
      default:
        begin
          \$1\result[2:0]  = 3'b011;
        end
    endcase
```

```
module test (in_a, in_b, out_vt);

input in_b;
input [1:0] in_a;
output out_vt;

reg   [2:0] result;
assign out_vt = result;

always @(*)
  begin
   if (in_b) begin
     case (in_a)
       2'b00 : begin
          result = 3'b101;  
       end

       2'b01: begin
          result    = 3'b001;
       end

       default result = 3'b011;

     endcase
   end
   else begin
     result = 3'b011;
   end
  end
endmodule
```

cliffordwolf, 2016-03-14:
Thanks for bringing this to my attention. I've now fixed the issue with duplicate defaults in commit 2a8d5e64f5a994fa8f4f51c00d647ad977e42e4b. I've also moved a very important warning from a comment in the verilog back-end sourcecode to the help message: Do not use the verilog backend on designs with unmapped RTLIL processes! There are some cases in which the generated always blocks have different simulation behavior than the original code, because RTLIL processes do not directly map to Verilog always blocks in all cases. You should at least run `proc` after running `hierarchy` and before running `write_verilog`.

closed 2016-03-14 by cliffordwolf