YosysHQ/yosys #127: Mishandling of a sized localparam bit selector

ravenexp, 2016-03-17:
The problem is demonstrated by the module below:

```
module test_genvar2localparam
        (
         output [3:0] d_o         
         );
        generate
                genvar   i;

                for (i = 0; i < 4; i = i + 1) begin : loop
                        localparam [1:0] di = i; // 0 1 2 3
                        assign d_o[di] = 1'b1;
                end
        endgenerate
endmodule
```

Expected result:

assign d_o = 4'b1111;

Actual result:

Warning: Range select out of bounds on signal `\d_o' at async_delay_line.v:11: Setting result bit to undef.
Warning: Ignoring assignment to constant bits at async_delay_line.v:11:
    old assignment: 1'x = 1'1
    new assignment: { } = { }.
Warning: Range select out of bounds on signal `\d_o' at async_delay_line.v:11: Setting 1 LSB bits to undef.
Warning: Ignoring assignment to constant bits at async_delay_line.v:11:
    old assignment: 1'x = 1'1
    new assignment: { } = { }.

ravenexp, 2016-03-17:
Synthesys output:

```
/* Generated by Yosys 0.6 (git sha1 UNKNOWN, x86_64-pc-linux-gnu-c++ 5.3.0 -O3 -DNDEBUG -march=native -O2 -fPIC) */

(* src = "async_delay_line.v:1" *)
module test_genvar2localparam(d_o);
  (* src = "async_delay_line.v:3" *)
  output [3:0] d_o;
  assign d_o[0] = 1'b1;
  assign d_o[1] = 1'b1;
  assign {  } = {  };
  assign {  } = {  };
endmodule
```

udif, 2016-03-17:
You are defining a localparam within the generate loop. Is that even legal?
Shouldn't the localparam assigned only once within the module?

cliffordwolf, 2016-03-17:
Simplified version of the test case:

```
module test(output [3:0] y);
        localparam [1:0] idx = 2;
        assign y[idx] = 1;
endmodule
```

The problem is that the localparam somehow inherits the signedness of the value. (Genvars are signed 32 bit integers, so are decimal integer constants.) So this effectively assigns to `y[-2]` (see `range=[-2:-2]` under the `AST_ASSIGN` node):

```
-- Running command `read_verilog -dump_ast2 test.v' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `test.v' to AST representation.
Generating RTLIL representation for module `\test'.
Dumping Verilog AST after simplification:
    AST_MODULE <test.v:1> [0x24fb640] str='\test'
      AST_WIRE <test.v:1> [0x24fb730] str='\y' output port=1 range=[3:0]
        AST_RANGE <test.v:1> [0x24fbaa0] range=[3:0]
          AST_CONSTANT <test.v:1> [0x24fb8a0] bits='00000000000000000000000000000011'(32) signed range=[31:0] int=3
          AST_CONSTANT <test.v:1> [0x24fb9a0] bits='00000000000000000000000000000000'(32) signed range=[31:0]
      AST_LOCALPARAM <test.v:2> [0x24efff0] str='\idx' range=[1:0]
        AST_CONSTANT <test.v:2> [0x24fcec0] bits='10'(2) range=[1:0] int=4294967294
        AST_RANGE <test.v:2> [0x24f0190] range=[1:0]
          AST_CONSTANT <test.v:2> [0x24f0260] bits='00000000000000000000000000000001'(32) signed range=[31:0] int=1
          AST_CONSTANT <test.v:2> [0x24f0330] bits='00000000000000000000000000000000'(32) signed range=[31:0]
      AST_ASSIGN <test.v:3> [0x24ef930]
        AST_IDENTIFIER <test.v:3> [0x24efc90 -> 0x24fb730] str='\y'
          AST_RANGE <test.v:3> [0x24efb90] range=[-2:-2]
            AST_CONSTANT <test.v:3> [0x24f00c0] bits='10'(2) range=[1:0] int=4294967294
        AST_CONSTANT <test.v:3> [0x24efd90] bits='00000000000000000000000000000001'(32) signed range=[31:0] int=1
--- END OF AST DUMP ---
Warning: Range select out of bounds on signal `\y' at test.v:3: Setting result bit to undef.
Warning: Ignoring assignment to constant bits at test.v:3:
    old assignment: 1'x = 1'1
    new assignment: { } = { }.
Successfully finished Verilog frontend.
```

commit ef4207d5 (2016-03-18): Fixed localparam signdness, fixes #127
closed 2016-03-18 by cliffordwolf