YosysHQ/yosys #129: ERROR: Assert `chunk.width >= 0' failed in kernel/rtlil.cc:3039.

olofk, 2016-03-18:
The following code triggers an assert. Changing DEPTH or any of the indices in the assignment seems to make the problem disappear

module a
  #(parameter DEPTH = 0)
   (input [1:0]      i);

   wire q = i[DEPTH-1:1];
   
endmodule

cliffordwolf, 2016-03-19:
Jfyi: After I have fixed it his will probably become an error.

If you have bogus default parameter values on your modules you might want to call `read_verilog` with `-defer` and make sure that you call `hierarchy` with `-top <top-module>` so the modules are never evaluated with their default parameters (unless they are instantiated this way somewhere in the design hierarchy).

olofk, 2016-03-19:
Ah ok. Is there any reason not to always use -defer?

cliffordwolf, 2016-03-19:
For straight forward synthesis flows you can make it the default using something like `verilog_defaults -add defer` (and maybe calling `verilog_defaults -clear` before running `synth_...`) . But in many formal flows you don't have a simple design hierarchy with a single top module, and I actually like to see errors from the Verilog/AST front-ends right after `read_verilog` and not much later after `hierarchy`.

Also: I think this way around the support overhead is smaller. Otherwise I'd have people asking why the modules don't show up after `read_verilog`.. Especially in cases where `read_verilog` is used later in the flow to load additional modules.

Last but not least it is the default for all front-ends to directly create the modules described in the input files. I think it would be a bit confusing if the verilog front-end would be different.

But the downside is of course that your default parameter values must make sense. I personally think they should anyways.. But there is an argument to be had here of course.

olofk, 2016-03-21:
Great. `verilog_defaults -add -defer` works fine for the FuseSoC-based flow.

cliffordwolf, 2016-08-21:
..turns out it would be pretty hard to give a nice error message for something like that in all cases, so I'm not going to invest any time on it since its just a "bad error message vs. good error message" problem.

closed 2016-08-21 by cliffordwolf

smoe, 2016-10-01:
Here another way to trigger the  "Assert `chunk.width >= 0' failed in kernel/rtlil.cc:3039." error. I just mention it since this issue is the google top hit for it and it seems like it does not necessarily have anything to do with parameters. Here, a rather educational example I tend to think, the result of ind+7 was not known at the time it was needed. For the typical software engineer this is somewhat counter-intuitive since the behaviour is asymmetric, i.e. the ind+7 on the left hand side of the '<=' is just fine.
```
module top(input iCE_CLK);

    reg [0:31] voutput=32'd0;

    reg [0:7] dummy1;
    reg [0:7] dummy2;
    reg [0:7] dummy3;
    reg [7:0] ind = 8'd0;
    reg [7:0] indplus7;

    always @(posedge iCE_CLK) begin
        voutput[ind:(ind+7)] <= dummy2; // ok

        voutput[0:7] <= dummy1;         // ok

        dummy2 <= voutput[ind:(ind+7)]; // error

        indplus7 = ind + 7;
        dummy3 <= voutput[ind:indplus7]; // ok
    end

endmodule
```

cliffordwolf, 2016-10-02:
@smoe just in case you are not aware: The code you posted is not legal Verilog. (See e.g. the errors iverilog produces when trying to parse that.) Part select expressions must be constant.

The standard conforming (and supported) way of doing what you are trying to do here is `voutput[ind +: 7]`.

smoe, 2016-10-02:
@cliffordwolf   This was unexpected. Just when I had thought to have understood something :) I was not aware and thank you tons for your correction. Back to (another) text book.