YosysHQ/yosys #131: fatal error: invalid .names directive

olofk, 2016-03-19:
I always get an error when running arachne-pnr. I've tried a few different designs, but they all fail on a line similar to:

.names $abc$172$n96 ready $abc$172$n84 $abc$172$n49

Not completely sure it's a yosys error, but I think it started to appear after upgrading to yosys bb9374b67c

cliffordwolf, 2016-03-20:
What is the Verilog input and Yosys script? I cannot reproduce this with any of the examples bundled with Project IceStorm or my PicoRV32 based demo SoC.

olofk, 2016-03-20:
[ico.zip](https://github.com/cliffordwolf/yosys/files/181281/ico.zip)

I have attached source code, makefile and the generated blif from yosys.

cliffordwolf, 2016-03-20:
When running `verilog_defaults -add -defer` before your `read_verilog` commands, add `verilog_defaults -clear` before running `synth_ice40`, or you will break the calls to the verilog front-end that happen from within `synth_ice40`:

```
verilog_defaults -add -defer
read_verilog ../src/ico/cnt.v
read_verilog ../src/ico/rot.v
verilog_defaults -clear
synth_ice40 -blif ico.blif -top top
```

Or, if you would like to write extra-clean code:

```
verilog_defaults -push
verilog_defaults -add -defer
read_verilog ../src/ico/cnt.v
read_verilog ../src/ico/rot.v
verilog_defaults -pop
synth_ice40 -blif ico.blif -top top
```

closed 2016-03-20 by cliffordwolf