YosysHQ/yosys #132: Override verilog parameters

olofk, 2016-03-21:
I can't find information on how to override verilog parameters with yosys. This is commonly used to set top-level parameters at synthesis-time.

cliffordwolf, 2016-03-21:
The command you are looking for is [`chparam`](http://www.clifford.at/yosys/cmd_chparam.html). However there was a problem with combining `chparam` and `read_verilog -defer`. I've now fixed that in commit 2c7e107d7a55a47fe6f6943e4541014649b57567.

Example Verilog code:

```
module submod(input [7:0] a, output [7:0] y);
  parameter integer increment = 0;
  initial begin
    if (increment == 0) begin
      $display("increment parameter must have a non-zero value!");
      $stop;
    end
  end
  assign y = a + increment;
endmodule

module topmod(input [7:0] a, output [7:0] y);
  parameter integer incr = 0;
  submod #(incr) submod_instance (a, y);
endmodule
```

Yosys script using `chparam` and `read_verilog -defer`:

```
read_verilog -defer test.v
chparam -set incr 42 topmod
synth -top topmod
```

Work-around for Yosys 0.6 (this also works with git head):

```
read_verilog -defer test.v
chparam -set incr 42 $abstract\topmod
synth -top topmod
```

closed 2016-03-21 by cliffordwolf

adumont, 2018-07-10:
I was facing a similar need. This question showed up while I was working on a solution. I ended up with a different approach though:

I'm creating a wrapper to my top module, written in m4 language. It's very simple, it overrides the parameters value, and then includes my top module definition.

Then in the Makefile, I process the wrapper.m4 file, to create the resulting wrapper.v file, that will be input to yosys.

I have detailled my approach [here](https://maker.itnerd.space/define-verilog-parameters-at-synthesis-time-yosys/), in case it can be useful anyone.