YosysHQ/yosys #133: ERROR: Abort in frontends/ast/genrtlil.cc:533.

AngelTerrones, 2016-03-23:
Version (Arch Linux x64):
`Yosys 0.6+8 (git sha1 d117893, clang 3.7.1 -march=x86-64 -mtune=generic -O2 -fstack-protector-strong -fPIC -Os)`

Im trying to parse/check some verilog file generated using MyHDL but I got the following warnings/errors:

`yosys CoreHDL.v`

```
...
Warning: Identifier `\COREHDL_CORE_ICACHE_LRU_M_STEP_1.expand' is implicitly declared at CoreHDL.v:2673.
Warning: Range select out of bounds on signal `\COREHDL_CORE_ICACHE_LRU_M_STEP_1.expand' at CoreHDL.v:2673: Setting result bit to undef.
Warning: Range select out of bounds on signal `\COREHDL_CORE_ICACHE_LRU_M_STEP_1.expand' at CoreHDL.v:2675: Setting result bit to undef.
Warning: Range select out of bounds on signal `\COREHDL_CORE_ICACHE_LRU_M_STEP_1.expand' at CoreHDL.v:2673: Setting result bit to undef.
Warning: Range select out of bounds on signal `\COREHDL_CORE_ICACHE_LRU_M_STEP_1.expand' at CoreHDL.v:2678: Setting result bit to undef.
Warning: Range select out of bounds on signal `\COREHDL_CORE_ICACHE_LRU_M_STEP_1.expand' at CoreHDL.v:2695: Setting result bit to undef.
Warning: Range select out of bounds on signal `\COREHDL_CORE_ICACHE_LRU_M_STEP_1.expand' at CoreHDL.v:2700: Setting result bit to undef.
Warning: Range select out of bounds on signal `\COREHDL_CORE_ICACHE_LRU_M_STEP_1.expand' at CoreHDL.v:2695: Setting result bit to undef.
Warning: Range select out of bounds on signal `\COREHDL_CORE_ICACHE_LRU_M_STEP_1.expand' at CoreHDL.v:2700: Setting result bit to undef.
Note: Assuming pure combinatorial block at CoreHDL.v:2662 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
ERROR: Abort in frontends/ast/genrtlil.cc:533.
```

A segment of the offending code:
```
always @(core_icache_current_lru, core_icache_access_lru) begin: COREHDL_CORE_ICACHE_LRU_M_STEP_1
    reg value;
    reg [1-1:0] tmp1;
    reg [2-1:0] expand [0:2-1];
    integer j;
    integer i;
    integer offset;
    reg [2-1:0] tmp0;
    
    offset = 0;
    for (i=0; i<2; i=i+1) begin
        expand[i][i] = 1'b1;
        for (j=(i + 1); j<2; j=j+1) begin
            expand[i][j] = core_icache_current_lru[((($signed({1'b0, offset}) + j) - i) - 1)];
        end
        for (j=0; j<i; j=j+1) begin
            expand[i][j] = (!expand[j][i]);
        end
        offset = (((offset + 2) - i) - 1);
    end
    tmp0 = 2'h0;
    for (i=0; i<2; i=i+1) begin
        value = 1'b1;
        for (j=0; j<2; j=j+1) begin
            value = (value && expand[i][j]);
        end
        tmp0[i] = value;
    end
    core_icache_lru_pre = tmp0;
    for (i=0; i<2; i=i+1) begin
        if (core_icache_access_lru[i]) begin
            for (j=0; j<2; j=j+1) begin
                if ((i != j)) begin
                    expand[i][j] = 1'b0;
                end
            end
            for (j=0; j<2; j=j+1) begin
                if ((i != j)) begin
                    expand[j][i] = 1'b1;
                end
            end
        end
    end
    offset = 0;
    tmp1 = 1'h0;
    for (i=0; i<2; i=i+1) begin
        for (j=(i + 1); j<2; j=j+1) begin
            tmp1[(((offset + j) - i) - 1)] = expand[i][j];
        end
        offset = (((offset + 2) - i) - 1);
    end
    core_icache_update_lru = tmp1;
end
```
The problem is, after some testings, the declaration of `reg [2-1:0] expand [0:2-1];` inside the `always` block. But the whole file, without changes compiles OK with Xilinx ISE 14.7 and iverilog.

Is this a yosys bug, or an invalid construct in Verilog?

cliffordwolf, 2016-04-18:
This seems to be a bug in Yosys. Here is a smaller test case (for the record):

```
module test(input [7:0] A, output reg [7:0] Y);
  always @* begin:myblock
    reg [7:0] myarray [0:1];
    myarray[0] = A + 23;
    myarray[1] = myarray[0] - 42;
    Y = myarray[1] + 19;
  end
endmodule
```

I'll look into it, but can't say yet when I will find the time to do so.

cliffordwolf, 2017-01-05:
Fixed in commit 81a9ee2. Sorry for the long delay.

closed 2017-01-05 by cliffordwolf