YosysHQ/yosys #146: Custom logic configuration or generating logic with equal delays?

cb88, 2016-04-05:
Is it possible to specify that certain logic should have equal delays?

It's been awhile since I've written any HDL but wouldn't the delay constraints in the generated BLIF files can define this? I guess that doesn't guarantee it can be synthesized but it should be possible to write code for this right?

Basically I'm wanting to try out writing some simple wave pipelines things and am not sure if yosys supports this.  Or would a wave pipeline be something that would need to be implemented as a black box to the HDL code and hand tuned by wiring up the LUTs manually (presuming your delay models are accurate enough for it to work to beging with)?

cliffordwolf, 2016-04-06:
>  delay constraints in the generated BLIF files

Yosys cannot read or generate BLIF files with delay constraints.

You can add delay constraints to the generated BLIF files, but I would not know which tool you'd use to further process those files then..

> wave pipelined things and am not sure if yosys supports this

There is no direct support for wave pipelining in yosys. Also, I'm not even sure what a such a support would look like. Considering where Yosys sits in a typical synthesis flow I'd expect dedicated support for something like that in the lower levels of the flow, where useful timing information is available, e.g. during things like gate sizing.

> For instance since there are no fast hard IP blocks for ALUs etc... it might pay off to have manually routed known fast implementation.

Are you talking about the iCE40 flow? (Because you don't mention iCE40 anywhere in your post, but I'd assume you are talking about a specific flow with that statement, because otherwise it wouldn't make any sense.)

Closing that issue because there is no bug report, specific feature request, or anything else that needs to be done.

closed 2016-04-06 by cliffordwolf

cb88, 2016-04-06:
ice40 of course I have the ice40 8k dev board. So, the lower level operations are more likely to be occurring in arachne-pnr then? I guess that makes sense since it would have the information on where to route things in the FPGA even if you were to give it a prerouted block of LUTS to place in the FPGA it is conceivable that it work with that.

Might have been better to have just emailed you about it but some open source devs I know dislike that for some reason. Thanks!