YosysHQ/yosys #148: Simulation mismatch RTL vs. Synthesized.

Kmanfi, 2016-04-08:
Copy all files from Archive.zip and simlib.v to same directory and execute ./st.sh
Iverilog and vsim both give same simulation output. b_out will change to 1 at 135.
Synthesized result gives different output. b_out doesn't change to 1.

Files:
SCRIPT: st.sh
RTL: stg_bug.v
TB: tb_stg.v
YS: stg.ys
NET: y_stg_bug.v (script will generate this)

[Archive.zip](https://github.com/cliffordwolf/yosys/files/210035/Archive.zip)

cliffordwolf, 2016-04-08:
From your test bench:

```
#3
in_l = 1'b0;      // !!!!!!!!!!!!!!!!!!!!!!
clk = 1'b1;
in_e = 1'b1;
in_l = 1'b1;
#3
```

Changing the clock and data lines that drives logic that is synchronous to the clock in the same simulation time step obviously has undefined behavior in this context.

Be sure to understand how the Verilog stratified event queue works when using coding styles like that. Alternatively just stick to either updating data lines at time steps that do not contain an active clock edge (posedge clk in your example), or use proper synchronization techniques to synchronize your data line updates with the clock, such as:

```
reg clk = 1;
always #5 clk = ~clk;

initial begin
  @(posedge clk);   // synchronize to clock event

  in_d <= 1'b0; // nonblocking assignments
  in_e <= 1'b0;
  in_l <= 1'b0;

  @(posedge clk);   // synchronize to next clock event

  in_d <= 1'b0; // nonblocking assignments
  in_e <= 1'b0;
  in_l <= 1'b0;

  ...
end
```

The famous Cliff Cummings paper "Nonblocking Assignments in Verilog Synthesis, Coding Styles That Kill!" contain a brief introduction to the Verilog stratified event queue: http://www.sunburst-design.com/papers/CummingsSNUG2000SJ_NBA.pdf

(Also see his paper "RTL Coding Styles That Yield Simulation and Synthesis Mismatches" for an incomplete list of other dangerous things every Verilog coder should be aware of: http://www.sunburst-design.com/papers/CummingsSNUG1999SJ_SynthMismatch.pdf)

mtuomi, 2016-04-08:
Hi @cliffordwolf we are working with @Kmanfi on a huge (~1M lines of verilog) silicon proven design. The design has a top level test bench that works nicely on modelsim. However we haven't got the design to work as yosys netlist. This particular case is one of the isolated differences between modelsim and yosys we have found. We do understand the event driven simulation and sensitivity list etc. It's difficult to isolate a "bug" case without the whole design. You're right, the syntax used in the tb_stg.v might not be the best, but the difference exists also with vcd input stimulus taken from full design simulation. Also with the full design testbench that reads input stimulus synchronously from stimulus file. 

We will try to narrow the case down even more and update.

cliffordwolf, 2016-04-08:
> but the difference exists also with vcd input stimulus taken from full design simulation

There is no bug in the code snippet you have posted. E.g. `bash tests/tools/autotest.sh stg_bug.v` runs a simulation between pre- and post-synthesis verilog for 1000 cycles with random inputs. Considering that your example has 1 bit of internal state and 4 non-clock inputs, this is an exhaustive search of the entire state space.

This test case is even small enough that you can easily see that the output is correct by simply comparing the output netlist manually to the HDL code.

I can't comment on the effects you observe in your full design, because I have not seen it. All I can tell you is that there is definitely no synthesis bug in your `stg_bug.v` file and the reason why you see the difference in the simulation is because you have a bug in your testbench, as I have pointed out. The way the testbench is written is *not* "not the best": It is wrong, it is erroneous, its the reason why you observe different outputs for pre- and post-synthesis netlists, not a bug in yosys.

I'm looking forward to a test case that actually isolates to bug you observe in your large design.

mtuomi, 2016-04-20:
Hi @cliffordwolf , I sent three cases that fail autotest.sh to your personal email on 12. May. Were you able to reproduce?

cliffordwolf, 2016-04-20:
Hmm. I'm assuming that is 12th Mar, not May.. I double checked my inbox and spam folder, but can't find anything. This is the clifford.at address? You are sure you did not receive a bounce email from your outgoing mail server? (A few mail servers do not deliver to clifford.at bc. the servers current ip address used to be on some spamer lists. But you should have received a bounce message from your mail server in that case.) Can you try send it again to cliffordvienna@gmail.com?

cliffordwolf, 2016-04-20:
Got the test cases. Reopening this issue as reminder that there is something todo.

cliffordwolf, 2016-04-21:
Fixed in commits 1761d08dd239bcf1765ebf807fc22514edac387f and 5a09fa45535ffceae90359be727d2ff6e0ac2c58.

Kmanfi, 2016-04-22:
Hi, I sent a new case to cliffordvienna@gmail.com.

cliffordwolf, 2016-04-22:
For the record in case anyone's reading along: That last one was not a synthesis bug.

cliffordwolf, 2016-04-25:
Got it and I can reproduce the problem. I'm not sure when I will get to it. Might take until the end of the week..

Kmanfi, 2016-04-26:
Should this pass with autotest.sh?
```
module zx_test( in, en, out);
  input [7:0] in;
  input en;
  inout [7:0] out;

  assign out = en ? in : 8'bx;
//  assign out = en ? in : 8'bz;
//  assign out = en ? in : 8'b0;
endmodule

```

cliffordwolf, 2016-04-26:
Oops. Yes it should! ;)

Thanks for spotting this! Fixed in commit 0d2923c.

cliffordwolf, 2016-04-26:
The test case from two days ago turned out to be a case where undefs propagate differently in verilog ieee-std conforming simulation and synthesis. (Essentially: An `if` is just a MUX in synthesis, as is `?:`, but in simulation `if (foo)` treats an undefined (aka `x`) value for `foo` like a false. Most verilog designers will have seen cases where designs reset properly in simulation *and sometimes also in hardware*, but in post-synthesis simulation the x's just keep floating around and just won't go away. That's usually something like this.) So no bug.

closed 2016-04-26 by cliffordwolf

udif, 2016-04-27:
Just to add, the coding guidelines I'm working with, requires using ' ? ...  :  ...' instead of 'if (...)' exactly for that reason (Blocking X propagation).

cliffordwolf, 2016-04-28:
@udif that sounds... reasonable, but horrible to work with. :) I can't imagine writing Verilog code without `if` and `case`. And it doesn't solve the problem entirely. E.g. the following two lines propagate an `x` in the signal `a` differently:

    assign y = a ? b+1 : c+1;
    assign y = (a ? b : c) + 1;

For a=1'bx, b=4b0000, c=4b0001 the first line will assign 4'b00xx and the second line will assign 4'bxxxx. However, most people would expect the expressions to be equivalent and might expect the synthesis tool to do this kind of transformation in one way or another.

One thing I have on my "maybe some time" todo list is writing an AppNote outlining all the possible problems with x-propagation in simulation vs. synthesis. Not sure if I'll ever do it. A year ago I posted [this](https://www.reddit.com/r/Verilog/comments/26a9re/differences_between_synthesis_and_simulation/) as a test and I did not enjoy the discussion that followed as much as I hoped I would..