YosysHQ/yosys #152: gcc compile 3 quibbles

react66, 2016-04-29:
Clifford great program, I compiled it because the precompiled one was too old.

On ubuntu 14.04 the GCC compile:

1) the deps for gcc compile are a shorter list than documented:
    bison flex libreadline-dev gawk
    tcl-dev libffi-dev git mercurial graphviz xdot pkg-config python3 

2) make config-gcc left the setting at the top of the makefile as
    config := clang     it should be
    config := gcc 

3) Like anybody who uses TCL, I have a .tclshrc file in my home directory,
    so when he does the following to find the version it screws up:

TCL_VERSION ?= tcl$(shell echo 'puts [info tclversion]' | tclsh)

a lot can happen when you start tclsh, I had to rename .tclshrc to get
around the problem.

cliffordwolf, 2016-04-29:
> 1) the deps for gcc compile are a shorter list than documented

The documented list of deps are for all configs, not just gcc. Btw, your list lacks `build-essential`, which is required for a gcc build (otherwise you won't have `gcc` or even `make`). So the only thing you left out from the original list is `clang`, which is kind of a no-brainer when you only do gcc builds..

> 2) make config-gcc left the setting at the top of the makefile as
config := clang it should be
config := gcc

No. it should not. See the generated `Makefile.conf` for the effective value.

Generally it is a good idea not to modify the Makefile itself for things like local configurations, so they are not accidentally committed to any version control system.

> [..] I have a .tclshrc file in my home directory, [..]

Thanks for pointing that out. I've changed the line to `TCL_VERSION ?= tcl$(shell bash -c "tclsh <(echo 'puts [info tclversion]')")` in commit 06d35ea, which fixes the issue.

closed 2016-04-29 by cliffordwolf

react66, 2016-04-29:
Wow thanks Clifford that's service.
    
    On 16-04-29 01:29 AM, Clifford Wolf
      wrote:
    
    
      
        1) the deps for gcc compile are a shorter list than
          documented
      
      The documented list of deps are for all configs, not just gcc.
        Btw, your list lacks build-essential, which is
        required for a gcc build (otherwise you won't have gcc
        or even make). So the only thing you left out from
        the original list is clang, which is kind of a
        no-brainer when you only do gcc builds..
      
        2) make config-gcc left the setting at the top of the
          makefile as
          config := clang it should be
          config := gcc
      
      No. it should not. See the generated Makefile.conf
        for the effective value.
      Generally it is a good idea not to modify the Makefile itself
        for things like local configurations, so they are not
        accidentally committed to any version control system.
      
        [..] I have a .tclshrc file in my home directory, [..]
      
      Thanks for pointing that out. I've changed the line to TCL_VERSION
          ?= tcl$(shell bash -c "tclsh <(echo 'puts [info
          tclversion]')") in commit 06d35ea, which fixes the
        issue.
      —
          it on GitHub