YosysHQ/yosys #155: test_autotb

Kmanfi, 2016-05-02:
If design contains modules with non valid parameter values, test_autotb might give false mismatches. 
Does test_autotb use -defer?

cliffordwolf, 2016-05-02:
test_autotb does not call read_verilog, so the question doesn't make much sense. autotest.sh does call read_verilog, but it creates test cases for each and every individual module in the input, so it would not make sense for it to call read_verilog with -defer as it would then need to instantiate all modules with its default parameters anyways, yielding exactly the same result.

It would be possible to add an option to autotest.sh to only consider one specific module (which would then be considered the "top" module of the hierarchy). Then we'd have a situation where it actually makes sense to use -defer. Otherwise I don't see how it could possible do anything meaningful.

Kmanfi, 2016-05-02:
Oops, I mixed test_autotb and autotest.sh.  "top" module option with -defer is what would do it.

closed 2016-05-02 by Kmanfi