YosysHQ/yosys #160: Synthesizing counter using LUT4 and CARRY

phanrahan, 2016-05-05:
I have been trying to synthesize a counter from low-level primitives,
and having trouble. Here is my verilog file.

module Add2 (input [1:0] I0, input [1:0] I1, output [1:0] O);
wire  inst0_O;
wire  inst1_CO;
wire  inst2_O;
wire  inst3_CO;
SB_LUT4 #(.LUT_INIT(16'hC33C)) inst0 (.I0(1'b0), .I1(I0[0]), .I2(I1[0]), .I3(1'b0), .O(inst0_O));
SB_CARRY inst1 (.I0(I0[0]), .I1(I1[0]), .CI(1'b0), .CO(inst1_CO));
SB_LUT4 #(.LUT_INIT(16'hC33C)) inst2 (.I0(1'b0), .I1(I0[1]), .I2(I1[1]), .I3(inst1_CO), .O(inst2_O));
SB_CARRY inst3 (.I0(I0[1]), .I1(I1[1]), .CI(inst1_CO), .CO(inst3_CO));
assign O = {inst2_O,inst0_O};
endmodule

module Register2CE (input [1:0] I, output [1:0] O, input  CLK, input  CE);
wire  inst0_Q;
wire  inst1_Q;
SB_DFFE #(.INIT(1'b0)) inst0 (.C(CLK), .E(CE), .D(I[0]), .Q(inst0_Q));
SB_DFFE #(.INIT(1'b0)) inst1 (.C(CLK), .E(CE), .D(I[1]), .Q(inst1_Q));
assign O = {inst1_Q,inst0_Q};
endmodule

module main (input  CLKIN, input  A, input  CE, output  D1);
wire [1:0] inst0_O;
wire [1:0] inst1_O;
Add2 inst0 (.I0(inst1_O), .I1({1'b0,A}), .O(inst0_O));
Register2CE inst1 (.I(inst0_O), .O(inst1_O), .CLK(CLKIN), .CE(CE));
assign D1 = inst1_O[1];
endmodule

This works. Note that I am reading an input A. If A is 1, then it counts. If A is 0, it doesn't count.

If I change the Add2 instance to use a constant (1) instead

Add2 inst0 (.I0(inst1_O), .I1({1'b0,1'b1}), .O(inst0_O));

it doesn't work. In fact, the whole adder simples to be eliminated. Here is the start of the resulting blif file.

# Generated by Yosys 0.6+      61 (git sha1 ce7c980, clang 7.3.0 -fPIC -Os)

.model main
.inputs CLKIN A CE
.outputs D1
.names $false
.names $true
1
.names $undef
.gate SB_DFFE C=CLKIN D=inst0.inst2_O E=CE Q=inst0.inst2_O
.attr src "count2.v:17"
.param INIT 0
.names inst0.inst2_O D1
1 1
.names $undef inst0.I0[0] 
...

I am not sure what is happening. It seems like the optimizer is a bit too aggressive. Any suggestions?

cliffordwolf, 2016-05-06:
Thanks for reporting this issue! This was a bug in `ice40_opt` that did not cause any problems with the circuits generated by Yosys itself, because those circuits just happened to use a LUT_INIT value that is invariant to reversing the order of input ports for all LUTs that are unmapped in `ice40_opt`.

It is great that you were able to isolate the issue to such a small test case (or just happened to find it there). It probably would have been a nightmare to isolate that bug in a large-scale design..

This is now fixed in commit 126da0ad3dfb7b75aaaadc97c7d40b495da9d164.

PS: There is no INIT parameter on SB_DFFE cells, because there is no hardware feature that allows setting the init value. In iCE40 all registers are always initialized to zero.

closed 2016-05-06 by cliffordwolf

phanrahan, 2016-05-06:
Works now. Thank you! And thanks for the note about the INIT parameter on DFF. 

yosys is a great tool.