YosysHQ/yosys #165: iverilog translate on/off with autotest.sh

Kmanfi, 2016-05-13:
Should this work with autotest.sh?
If I comment assign out then it works. It seems iverilog simulates code between translate_off/on.

```
module z_test( in, en, out);
  input [7:0] in;
  input en;
  inout [7:0] out;

//  assign out = en ? in : 8'bx;

  // synopsys translate_off

  assign out = 8'b10101010;

  // synopsys translate_on

endmodule

```

udif, 2016-05-13:
As it should.

'// synopsys ' type comments are old, unofficial way used by synopsys to prevent non-synthesizable code from being synthesized by its DC synthesis tool.
The intent was for you to put all your simulation-only code in those comments.

This approach had several problems:
1. This was non standard - nothin in the verilog standard mentions any "magic" keywords that may appear in a comment section and affect the behavior of a standard-compliant tool.
2. Even if we adopt the notion of using magic keywords in comments, why use a vendor keyword? So later, this was replaced by '// synthesis on/off'. (I don't remember who did this).
3. If you want to use comments for passing optional features which shouldn't affect the behavior , you should use the attriute mechanism - '(* ... *)'.

But generally, I think the best wayto handle this kind of issues is to use the `ifdef/`endif` mechanism, that is required by the standard.

cliffordwolf, 2016-05-14:
Yosys does support `// synopsys translate_off` ... `// synopsys translate_on` blocks. I.e. it will ignore everything within the block (and print a warning message saying you should use \`ifdef instead).

So this is expected not to work, because iverilog (as a simulator) does simply ignore those comments. (And therefore does not ignore the code between those comments.)

closed 2016-05-14 by cliffordwolf