YosysHQ/yosys #168: Width of memory access detection depends on literal

torgny, 2016-05-17:
Yosys 0.6 (git sha1 124f7c4) fails to detect the width of memory access in the following code
```
module m ();
  wire [0:0] foo[0:0];
  wire [0:0] bar[0:0];
  wire baz;

  assign foo[0] = {1{1'b0}};  // Replace with "assign foo[0] = 1'b0;" and yosys accepts the code
  assign bar[0] = 1'b0;
  assign baz = bar[foo[0]];
endmodule
```

If the assignment to foo is replace with the code in the comment, yosys manages to detect the width of the memory access. Expected behavior is that both of these variants of the above code works.

By the way, yosys 0.6 prints "Failed to detect with of memory access", that is, "with" instead of "width".

cliffordwolf, 2016-05-27:
This is a combination of two bugs. First, there is a access-after-delete bug in mem2reg that I have fixed now, and second there is a problem when mem2reg is active on a memories and they are "cascaded" for memory access as in `bar[foo[0]]`.

I've now fixed the access-after-delete bug (and the typos in the error message). Now you should reproducibly get a `ERROR: Failed to resolve identifier \foo for width detection at test.v:10!` error message. (That `assign foo[0] = 1'b0;` made it work was just a side effect of the memory bug.) The "failed to detect.." error message was just a result of the code reading uninitialized data, the "failed to resolve.." error message is "more correct" and will go away once I have fixed that second bug. 

I hope to find the time to look into the second bug soon.

cliffordwolf, 2016-08-21:
This is now fixed with current git head. Thanks for your patience.

closed 2016-08-21 by cliffordwolf