YosysHQ/yosys #169: ngspice SEGFAULTS on seeing `=' in .SUBCKT name

klammerj, 2016-05-18:
This seems the bit of spice that causes it:
```
.SUBCKT _paramod_ramp_bits=5 clk rst ctr.0 ctr.1 ctr.2 ctr.3 ctr.4
X0 ctr.0 1.0 NOT
X1 ctr.1 ctr.0 2 NAND
X2 2 3 NOT
X3 ctr.1 ctr.0 4 NOR
X4 4 3 1.1 NOR
X5 ctr.2 5 NOT
X6 2 5 6 NOR
X7 3 ctr.2 7 NOR
X8 7 6 1.2 NOR
X9 6 ctr.3 8 NOR
X10 ctr.3 ctr.2 9 NAND
X11 9 2 10 NOR
X12 10 8 1.3 NOR
X13 10 ctr.4 11 NOR
X14 ctr.4 12 NOT
X15 6 ctr.3 13 NAND
X16 13 12 14 NOR
X17 14 11 1.4 NOR
X18 clk 1.0 ctr.0 0s rst DFFSR
X19 clk 1.1 ctr.1 0s rst DFFSR
X20 clk 1.2 ctr.2 0s rst DFFSR
X21 clk 1.3 ctr.3 0s rst DFFSR
X22 clk 1.4 ctr.4 0s rst DFFSR
.ENDS _paramod_ramp_bits_5
```
(it originates from a parameterized verilog module...)
changing to:
.SUBCKT _paramod_ramp_bits_5 clk rst ctr.0 ctr.1 ctr.2 ctr.3 ctr.4
...works

here's some gdb output...
```
Program received signal SIGSEGV, Segmentation fault.
0xb79741c9 in __GI_strcmp (p1=0x0, p2=0xbfffdec7 "not") at strcmp.c:34
34      strcmp.c: No such file or directory.
(gdb) bt
#0  0xb79741c9 in __GI_strcmp (p1=0x0, p2=0xbfffdec7 "not") at strcmp.c:34
#1  0x08091655 in numnodes (name=0x87b6638 "x0 clk inv not", subs=0x87b6560, modnames=0x87b5f68)
    at ../../../src/frontend/subckt.c:1486
#2  0x08090ce8 in translate (deck=0x8789620,.
    formal=0x87b6508 "clk inv nrst cmp1 bits1 lo1 hi1 cmp2 bits2 lo2 hi2 cmp3 bits3 hi3 cmp4 bits4 hi4",.
[...]

(gdb) frame 1
#1  0x08091655 in numnodes (name=0x87b6638 "x0 clk inv not", subs=0x87b6560, modnames=0x87b5f68)
    at ../../../src/frontend/subckt.c:1486
1486                if (eq(sss->su_name, s))
(gdb) list 1486
1481                *s-- = '\0';
1482            while ((*s != ' ') && (*s != '\t'))
1483                s--;
1484            s++;
1485            for (sss = subs; sss; sss = sss->su_next)
1486                if (eq(sss->su_name, s))
1487                    return (sss->su_numargs);
1488            /*
1489             * number of nodes not known so far.
1490             * lets count the nodes ourselves,
(gdb) p subs
$1 = (struct subs *) 0x87b6560
(gdb) p *subs
$2 = {su_name = 0x87b6578 "s_d", su_args = 0x87b6588 "clk rst cmp bit q.0 q.1 q.2 q.3 q.4 q.5",.
  su_numargs = 10, su_def = 0x8789658, su_next = 0x87ad898}
(gdb) p *sss
$3 = {su_name = 0x0, su_args = 0x87b64b8 "", su_numargs = 0, su_def = 0x879b148, su_next = 0x87ad288}
(gdb) p *subs->su_next
$4 = {su_name = 0x87ad3f0 "ramp", su_args = 0x87ad8b0 "clk rst ctr.0 ctr.1 ctr.2 ctr.3 ctr.4 ctr.5",.
  su_numargs = 8, su_def = 0x8788ee8, su_next = 0x87ad3c8}
(gdb) p *subs->su_next->su_next
$5 = {su_name = 0x87ad3e0 "dtg", su_args = 0x87ad880 "clk rst d a b", su_numargs = 5, su_def = 0x8788ca8,.
  su_next = 0x87ad3b0}
(gdb) p *subs->su_next->su_next->su_next
$6 = {su_name = 0x87ad5a0 "a",.
  su_args = 0x87b6508 "clk inv nrst cmp1 bits1 lo1 hi1 cmp2 bits2 lo2 hi2 cmp3 bits3 hi3 cmp4 bits4 hi4",.
  su_numargs = 17, su_def = 0x8787178, su_next = 0x87ad398}
(gdb) p *subs->su_next->su_next->su_next->su_next
$7 = {su_name = 0x0, su_args = 0x87b64b8 "", su_numargs = 0, su_def = 0x879b148, su_next = 0x87ad288}

[...]

(gdb) p *sss->su_def
$16 = {li_linenum = 55, li_linenum_orig = 0, li_line = 0x87ad5b0 "*param  _paramod_ramp_bits=5",.
  li_error = 0x0, li_next = 0x8786ba8, li_actual = 0x0}

```

cliffordwolf, 2016-05-20:
Not sure what to do with the ngspice stack trace. If you think this is a ngspice bug you should probably report it to the ngspice team..

However, in commit f3983a0 I've now added `=` to the list of characters that should be converted to `_` when writing a spice netlist. That should work around that problem.

closed 2016-05-20 by cliffordwolf

klammerj, 2016-05-20:
There does not seem to be any documentation in their ngspice.pdf manual about what constitutes a valid subcircuit identifier, so I can't really tell either.....
Their parser is handwritten spaghetti-style and there seems preprocessing to be going on in there. 
they allow key=value pairs in subckt definitions, and I think that part gets it wrong...
```
2.8.3 Subcircuit parameters
General form:
.subckt <identn> node node . . . <ident>=<value> <ident>=<value> . . .
```