YosysHQ/yosys #175: Autotest.sh and setundef -zero

Kmanfi, 2016-06-01:
When added `setundef -zero` to generation of _syn0.v, autotest.sh give ERROR with the code below. Is it ok?

I changed this line in autotest.sh 
`test_passes -f "$frontend" -p "hierarchy; proc; opt; memory; opt; fsm; opt -full -fine" $fn`
to
`test_passes -f "$frontend" -p "hierarchy; proc; opt; memory; opt; fsm; setundef -zero; opt -full -fine; setundef -zero; opt -full -fine" $fn`

Then tried it with this code.
```
module cg (    
    C,
    GC,
    E
);

    input  C;
    output GC;
    input  E;
    
    reg ED;

always@(*) begin
    if(~C)
        ED = E;
end

    assign GC = C & ED;

endmodule
```

cliffordwolf, 2016-06-01:
That's a race condition in simulation for the generated code. That the version without `setundef -zero` worked was pure luck. Fixed in commit adfc80727c19021f911833305f5d322af6b8a0aa.

Kmanfi, 2016-06-02:
I sent new code to you. Run with same setundef modification.

Kmanfi, 2016-06-02:
I sent another example to run without setundef modifications.

cliffordwolf, 2016-06-02:
Commit d2695e2 fixes the first test case you sent me. The 2nd one seems to be completely unrelated, as it does not even contain d-latches.

Unfortunately the 2nd test case is >16000 lines and I don't have much time to spare to look into it right now. I would assume however that this case is related to the standardized simulation-synthesis mismatch in `if()` and `case()` handling of undef values.

closed 2016-06-02 by cliffordwolf

Kmanfi, 2016-06-02:
About 2nd test case. I found flip flops without reset so, there are some issues in higher level. I continue with it.

Kmanfi, 2016-06-08:
I sent new code to you. Only few lines of code.