YosysHQ/yosys #177: $assert cells should be ignored during synthesis

whitequark, 2016-06-10:
This was triaged from openfpga, so please see the discussion over at https://github.com/azonenberg/openfpga/issues/11.

cliffordwolf, 2016-06-12:
The usual approach is to enclose all code for formal verification with \`ifdef FORMAL ... \`endif and only set the define FORMAL when doing formal verification (done by default by `read_verilog -formal`).

It would be possible, as @azonenberg suggested, to add an option to `synth_greenpak4` that simply runs a command such as `delete t:$assert` for convenience, but that would be very unconventional. I'd accept such a patch for `synth_greenpak4` from @azonenberg, but I would not add it to "my" synthesis flows such as `synth_ice40` or `synth_xilinx`.

closed 2016-06-12 by cliffordwolf

antonblanchard, 2020-05-18:
Thanks, I guess the SystemVerilog reproducer was a bit of a red herring. Should GHDL not emit Assert entries at all, or have an option to not emit them? It seems a bit strange to be doing it there, vs yosys or nextpnr.