YosysHQ/yosys #178: Feature request: option to give seed value for autotest.sh

Kmanfi, 2016-06-10:
I have situation where unit within part of bigger design will fail during autotest.sh.
When I isolate this unit as it's own it works because of different seed.
It would be nice to have possibility to give seed value for autotest.sh.

cliffordwolf, 2016-07-13:
The seed value is the current unix time stamp. This is the relevant code from `passes/tests/test_autotb.cc`:

        f << stringf("reg [31:0] xorshift128_w = %u; // <-- seed value\n", int(time(NULL)));

So just re-running autotest.sh should run the test with a different seed value.

Closing the issue. Please comment if this is not sufficient and you really need a way to set the seed value explicitly.

udif, 2016-07-13:
I think @Kmanfi wanted exactly the opposite, the ability to repay a specific seed so that seed-dependant failures can be recreated and regressed after a candidate fix was applied.

cliffordwolf, 2016-08-06:
Commit 88a67af adds `autotest.sh -S <seed_val>`.

closed 2016-08-06 by cliffordwolf