YosysHQ/yosys #179: iopadmap should propagate src attribute

whitequark, 2016-06-10:
Right now iopadmap-created cells have no location info whatsoever, which makes it impossible to report meaningful errors in PAR.

cliffordwolf, 2016-08-06:
Fixed in 90c17aa. (Does not set any attribute on the cell, but preserves the attribute on the wire.)

closed 2016-08-06 by cliffordwolf

whitequark, 2016-08-06:
Hm, this broke gp4par and after looking closer I do think it would make more sense to set all of the attributes on the cell. Currently, gp4par looks at the LOC attribute on the wire to determine where to place the GP_IOBUF primitive (https://github.com/azonenberg/openfpga/blob/master/src/greenpak4/Greenpak4NetlistCell.cpp#L45). This results in a bunch of weird and fragile code; in this case it broke because after your change, the LOC attribute was also set on the OUT port of some GP_2LUTs, which is of course nonsensical.

On top of that this change is completely unhelpful towards my original issue; before this I could have found the src attribute by looking at the wires of iopadmapped GP_IOBUF cell (which I didn't realize at the time of filing), and after this... I have no other choice!

There's one more problem, which is the fact that a PAR has to deal with inconsistency between manually instantiated iobuf cells and iopadmap-created ones, which affects both src and LOC attributes in case of gp4par; when manually instantiated it should look at the cell but when mapped it should look at the wires. I think PARs shouldn't be required to do this.

Is there some reason you don't want to copy the wire attributes to the cell?

cliffordwolf, 2016-08-07:
>  it would make more sense to set all of the attributes on the cell

We don't do this with any other cell types. wire attributes usually stay wire attributes.

Consider that cell attributes are potentially short-lived because of the way a group of cells can be replaced by a single cell or vice versa during the different stages of a synthesis flow.

You can't e.g. take a high level multiplier cell, set the "keep" attribute on it, and expect it to survive the synthesis flow as-is and get a netlist that actually makes sense for the target architecture. But you can do that with a wire. That's why usually attributes are defined on wires and stay on wires through the entire flow.

> in this case it broke because after your change, the LOC attribute was also set on the OUT port of some GP_2LUTs, which is of course nonsensical

If you think the current behavior is incorrect, please post a (short) example code demonstrating the issue.

> On top of that this change is completely unhelpful towards my original issue; before this I could have found the src attribute

Before this change the src attribute (like any other attribute) was removed from the wire. Therefore I don't understand what you are talking about: How where you able to find an attribute in the output netlist that was removed by the iopadmap command? Please post example verilog code (+yosys script) that demonstrates the issue.

> Is there some reason you don't want to copy the wire attributes to the cell?

First, the reasons stated above. Second, that usually the effect of an attribute like "keep" is different for wires and cells, and therefore it is generally a bad idea to simply blindly copy (all) wire attributes onto a cell.

However, it would certainly be possible to write a _separate_ command that has nothing to do with iopadmap for moving and/or copying (a user-specified selection of) attributes from wires to the cell(s) driving the wire. There are certainly cases where this could be useful.

But simply moving all attributes from the wire to the newly generated cell in iopadmap is just a really bad idea for so many reasons.

whitequark, 2016-08-07:
> That's why usually attributes are defined on wires and stay on wires through the entire flow.

Hm, yes, I did not fully understand this. I now agree that copying attributes from wires to cells in general would be a mistake. The iopadmap case is a bit more complex though.

> How where you able to find an attribute in the output netlist that was removed by the iopadmap command? Please post example verilog code (+yosys script) that demonstrates the issue.

> > in this case it broke because after your change, the LOC attribute was also set on the OUT port of some GP_2LUTs, which is of course nonsensical
> 
> If you think the current behavior is incorrect, please post a (short) example code demonstrating the issue.

Sure. I'll answer both questions at once. Let's take this code:
```verilog
module top(
        (* LOC="P2" *) input a,
        (* LOC="P3" *) input b,
        (* LOC="P4" *) output y,
    );
    assign y = a | b;
endmodule
```
and build it as `yosys foo.v -p "synth_greenpak4 -json foo.json"`. Before the iopadmap change, it results in the following output: [foo.json.txt](https://github.com/cliffordwolf/yosys/files/405606/foo.json.txt). As you can see, the attributes are set on module ports and nothing else. Especially, the nets connected to the GP_2LUT instance do not have any attributes on them. After the iopadmap change, the output is as follows:  [foo.json.txt](https://github.com/cliffordwolf/yosys/files/405609/foo.json.txt). The attributes are also set on every wire leading to the GP_2LUT cell, confusing the PAR.

Now let's talk about attributes on wires and cells specifically in the case of iopadmap. I will use the GreenPAK attributes but these aren't too specific to gp4par.
  * The `src` attribute. I think this one clearly can be copied from the wire onto the cell in the iopadmap pass. It is very useful to always have a `src` attribute on a cell that indicates where the cell comes from for error messages. I/O pads are currently the only case I'm aware of when that's not done, and this was the original motivation for this.
  * The `LOC` attribute. This attribute is, in fact, almost nonsensical on a wire, because its only purpose is to constrain the placement of a cell. This is how it works for every other cell, and again, I/O pads here are the exception, warranting the `FindLOC()` hack that I linked above.
  * The `PULLUP`, `PULLDOWN`, `DRIVE_STRENGTH`, etc attributes. These make sense to be set on the net, I think, Verilog even already has drive strengths on wires. They also don't cause any problems for PAR.
  * The `SCHMITT_TRIGGER`, `IBUF_TYPE`, etc attributes. These would perhaps make the most sense as primitive parameters, but like in the previous case, there's no rush to change this, so let's ignore them for now.

This leaves just two attributes, `src` and `LOC`. Given the universal usefulness of `src`, I think it should be copied by iopadmap in all cases, in the same way as other passes would propagate this attribute. As for the GreenPAK-specific `LOC`, specifying it on the module port is an important convenience, but otherwise it doesn't fit well. For it, I propose a new option `-moveattr` for the iopadmap pass, that would move (moving lets the PAR reject the `LOC` attribute if it is ever set on a net, as additional sanity checking) the specified set of attributes from the net corresponding to the module port, to the created cell.

cliffordwolf, 2016-08-08:
> Let's take this code [..]

Ok, I see. The problem is that my bugfix didn't fix a bug. It was ok before and now I replicate the attribute. (For some reason I thought the addWire(ID, other) would not copy the attributes.) So I've now undone that change to restore the original behavior (9d15529). Sorry I broke your flow..

Regarding the rest of your post: I have already offered to create a dedicated pass to move or copy attributes from wires to cells, that can of course be used on the iopad cells after running iopadmap. You are now proposing to add dedicated flags to add this feature to iopadmap only. I find this a very strange approach. Are you suggesting we are to replicate this feature in all other passes where this can be used as well? For example in shregmap?

I can't understand why you insist that this must be a feature built-in to iopadmap. Maybe you can elaborate in this?

whitequark, 2016-08-08:
> I can't understand why you insist that this must be a feature built-in to iopadmap. Maybe you can elaborate in this?

I don't have a preference for this feature being built-in to iopadmap. By all means, if you think a separate pass is the right choice for Yosys, I have no objections. It is simply that I can't imagine how would it be useful anywhere else, and so special-casing it in iopadmap seemed like less work.

To elaborate, every GP4 design (and probably most of them even once we have support for external constraint files) comes with the LOC attribute on module ports, simply because it is short and convenient, compared to instantiating buffers, or indeed external constraint files. Whereas, for anything else, once one needs device-specific attributes on some cell, it is exceedingly unlikely that one would be still interested in letting the frontend decide to create the cell on its own, like via shregmap, since the predictability that comes with instantiating the cell explicitly is more valuable than saving a few lines.

cliffordwolf, 2016-08-09:
Commit 3c6d31f adds the "attrmvcp" pass. Usage example:

    attrmvcp -attr LOC t:GP_OBUF n:*
    attrmvcp -driven -attr LOC t:GP_IBUF n:*

whitequark, 2016-08-09:
Thank you very much! I'll test it soon.

whitequark, 2016-08-09:
What about the `src` attribute? It could be done with the new pass, but don't other passes propagate it on their own?

cliffordwolf, 2016-08-09:
not sure what you mean by that. techmap and simplemap copy it from the cells they replace to the cells they replace it with. But they also change the src attribute (add new elements to it). But I don't understand the relevance in this context here.

There is nothing that prevents you from using attrmvcp on `src` attributes.

whitequark, 2016-08-09:
> But I don't understand the relevance in this context here.

I was thinking that it is generally useful for iopadmap to copy the src attribute itself.

> There is nothing that prevents you from using attrmvcp on src attributes.

I understand that. I'll just do that for synth_greenpak4 alone, sure.