YosysHQ/yosys #185: autotest.sh and timescale

Kmanfi, 2016-07-01:
Go to: 
`yosys_bigsim/aes_5cycle_2stage/rtl`
run:
`autotest.sh -p 'prep -ifx' aes_sbox.v`

Kmanfi, 2016-07-01:
Go to:
`yosys_bigsim/amber23/rtl`
run:
`autotest.sh -p 'prep -ifx' a23_coprocessor.v`

Kmanfi, 2016-07-01:
Go to:
`yosys_bigsim/amber23/rtl`
run:
`autotest.sh -p 'prep -ifx' a23_multiply.v`

cliffordwolf, 2016-07-02:
Commit e420412 fixes the \`timescale issue. I still have to look into the two amber23 test cases.

cliffordwolf, 2016-07-02:
`a23_coprocessor.v` and `a23_multiply.v` both require some regs to be initialized so they are not stuck in an undef state. (E.g. `count` and `product` in `a23_multiply.v`.) The netlist representation usually generated by the verilog back-end in `-noexpr` mode (the default for `autotest.sh`) does not preserve init values (they are stored on an attribute on the wire driven by the FF and have no effect on the simulation). Running `autotest.sh` with `-e` disables `-noexpr`, thus enables modelling of init values and fixes the problem with the two amber23 modules.

closed 2016-07-02 by cliffordwolf